lvgl/lvgl #7592: get idle for phtread

anaGrad, 2025-01-13:
### Introduce the problem

I was wondering if it would be possible to create a function similar to lv_os_get_idle_percent() from freertos:
https://github.com/lvgl/lvgl/blob/d9677acd93f6eeeb9f50f68363516dfae2233570/src/osal/lv_freertos.c#L408
but make it in lv_pthread.c for linux?

xlbao123, 2025-01-15:
we can read /proc/stat to calculate the CPU idle percentage directly in linux

kisvegabor, 2025-01-16:
I was thinking about the same in the past days. We can use `/proc/self/stat` to get the CPU usage of the current process.

We can make `lv_os_get_idle_percent` mandatory and in the OS implementations. Where there is no proper implementation for it (and in case of no OS) we can use `lv_timer_get_idle` as a fallback. This way we can set `LV_SYSMON_GET_IDLE lv_os_get_idle_percent` by default. 

What do you think?

kisvegabor, 2025-01-16:
@liamHowatt just pointed out on Slack that we should monitor the system's CPU usage, not LVGL's CPU usage. but that also doable via `/proc/stat`.

nicusorcitu, 2025-01-17:
I agree with @kisvegabor suggestion.

AndreCostaaa, 2025-01-17:
The idle time can also be found via `/proc/uptime`

From the man page:
```
NAME
       /proc/uptime - system uptime

DESCRIPTION
       /proc/uptime
              This file contains two numbers (values in seconds): the uptime of the system (including time spent in suspend) and the amount of time spent in the idle process.
```

nicusorcitu, 2025-01-20:
I believe that is the total system uptime from the boot moment.

AndreCostaaa, 2025-01-20:
Doesn't `proc/stat` also give you the total time since system boot?

The solution would be to store the stats when the app launches and calculate the difference as the program runs

anaGrad, 2025-01-20:
Hi @kisvegabor ! Is there a plan for implementing this on your side?

AndreCostaaa, 2025-01-20:
Would you like to try #7632 ? @anaGrad

anaGrad, 2025-02-04:
Hi @AndreCostaaa ! Sorry for the long reply. I finally got a chance to test your PR and I can say it works perfect. Thank you!

PR #7632 (2025-02-14): feat(osal): add lv_os_get_idle_percent for linux
Fixes #7592 

- Implemented lv_os_get_idle_percent for linux by reading the `proc/uptime` file which returns the active and idle times of the system since boot. From `man proc_uptime`:

```
proc_uptime(5)                                                                                   File Formats Manual                                                                                  proc_uptime(5)

NAME
       /proc/uptime - system uptime

DESCRIPTION
       /proc/uptime
              This file contains two numbers (values in seconds): the upt [... cut]
closed 2025-02-14 by FASTSHIFT