lvgl/lvgl #7600: When using lv_menu_create(parent, title) the parent created for the page is not the parent defined in this function call.

phelter, 2025-01-13:
### LVGL version

v9.2

### Platform

Visual Studio simulator on Linux.

### What happened?

The definition of:

https://docs.lvgl.io/master/API/widgets/menu/lv_menu.html#_CPPv419lv_menu_page_createP8lv_obj_tPCKc

Describes a ` newPage = lv_menu_page_create(parent, name)  - but when using the lv_obj_get_parent(newPage) this provides an interim object who's parent is not the parent expected - in other words `parent != lv_obj_get_parent(newPage)` - Please fix your documentation to identify that there might be intermediary objects between the menu object and the menu page object OR fix it so that there is no intermediary.

Reason is I'm encapsulating this in C++ and instead of having the need to decide what the menu object is or storing this per page, I want a function associated with the page object to display - which is equivalent to what I thought would be:

```cpp
lv_menu_set_page(lv_obj_get_parent(m_obj), m_obj);
```

but what it in fact is:
```cpp
lv_menu_set_page(lv_obj_get_parent(lv_obj_get_parent(m_obj)), m_obj);
```

Also if there are any other objects that are created with intermediary objects can those also be documented.

### How to reproduce?

```cpp
auto menu = lv_menu_create(screen);
auto menu_page = lv_menu_page_create(menu, "top");
assert(menu == lv_obj_get_parent(menuPage));
```

vwheeler63, 2025-01-14:
@kisvegabor This is a great example of the usefulness of having the run-time object hierarchy (that we have discussed) illustrated in the documentation for each widget that has many parts....  Also, as I try to learn how to use `lv_menu_t` via its documentation, I have again run into that barrier I have mentioned before, and conclude that it really does need an overhaul (deep-dive into the C code) because the original writer (as far as I can tell) did not record how he intended Menu widgets to be used.  I could not find traces of that intention in the `menu.rst` or in the API documentation in `lv_menu.h`.  The only hint I found was in a repeating pattern in the examples.  Frustrating.  I'm going to vote to add Menu documentation to the "full overhaul" list.

vwheeler63, 2025-01-14:
This should be handled by PR #7604.  @phelter do you concur?

phelter, 2025-01-14:
@vwheeler63 Unfortunately no - the documentation in #7604 describes the menu create hierarchy which is great info and very helpful for understanding.  This is issue is to tackle the hierarchy of the menu_page object that you get back from `lv_menu_page_create()` - which is based on the other documentation something like:

```markdown
This creates a Menu Widget with this object hierarchy:
    - Menu Page (Wrapper): :cpp:type:`lv_obj_t`
        - Menu Page: :cpp:type:`lv_menu_page_t`
```

liamHowatt, 2025-01-14:
@vwheeler63 see also #7586

vwheeler63, 2025-01-14:
> @vwheeler63 see also #7586

Nice!  Thank you!

PR #7604 (2025-01-30): fix(docs):  clarify obj hierarchy after `lv_menu_page_create()`
In Issue #7600, @phelter correctly pointed out that the API documentation for `lv_menu_page_create()` (and its 1st argument name) is misleading about the resulting object hierarchy after the call.  This submission both clarifies the result in the API documentation as well as adds some enhancements to `menu.rst`, though it still needs a thorough review.

Resolves #7600

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.  Done.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/ex [... cut]
closed 2025-01-30 by kisvegabor