lvgl/lvgl #7601: define LV_FS_WIN32_LETTER "\0" error when use LV_USE_TJPGD

USTHzhanglu, 2025-01-14:
### LVGL version

v9.0.0

### Platform

windows SDL2 with vscode

### What happened?

```
lv_image_set_src: failed to get image info: 0x7ff7687c20e0 lv_image.c:175
```
and no img show
change `\0` -> `A` ok
if path error it always show str
```
lv_image_set_src: failed to get image info: A:test.jpeg lv_image.c:175
```

### How to reproduce?

```c
lv_image_set_src(image1, "\0:test.jpeg");
```

USTHzhanglu, 2025-01-14:
`lv_image_src_get_type` don`t think `\0` is file path.
```c
    /*The first byte shows the type of the image source*/
    if(u8_p[0] >= 0x20 && u8_p[0] <= 0x7F) {
        return LV_IMAGE_SRC_FILE; /*If it's an ASCII character then it's file name*/
    }
    else if(u8_p[0] >= 0x80) {
        return LV_IMAGE_SRC_SYMBOL; /*Symbols begins after 0x7F*/
    }
    else {
        return LV_IMAGE_SRC_VARIABLE; /*`lv_image_dsc_t` is draw to the first byte < 0x20*/
    }
```
but `lv_conf_template.h` make default LETTER as `\0`. if I use it directly, it will error.
maybe change the `lv_conf_template.h`?

vwheeler63, 2025-01-14:
Actually, I think the error is in the comment in `lv_conf_template.h`.  It is my understanding that if one uses `lv_fs_win32.c` or anything that uses it (e.g. File Explorer), that it is a requirement of the programmer to change '\0' to a driver identifier of their choosing.  However, given the way that macro is used (to provide a default value for the `driver.letter` field), I find that the instruction in the comment for this macro is misleading.  Currently it is:

```c
/**< Set an upper cased letter on which the drive will accessible (e.g. 'A') */
```

but it seems to me that it should be
```c
/**< Set an upper cased letter to be the identifier for this file-system driver (e.g. 'A').  See docs under File System > Identifier Letters. */
```

Once done, then the call above that causes it should be (assuming 'A' as the driver identifier):
```c
lv_image_set_src(image1, "A:C:/path/to/test.jpeg");  /* If drive and directory should be specified. *
lv_image_set_src(image1, "A:test.jpeg");  /* If default drive and directory are already correct. *
```

@liamHowatt , what do you think?

liamHowatt, 2025-01-16:
It was fixed by #6367 (and improved by #6986). Are you able to use a newer version of LVGL? It generates a compile-time error if a driver letter is `'\0'` now.

USTHzhanglu, 2025-01-20:
> It was fixed by [#6367](https://github.com/lvgl/lvgl/pull/6367) (and improved by [#6986](https://github.com/lvgl/lvgl/pull/6986)). Are you able to use a newer version of LVGL? It generates a compile-time error if a driver letter is `'\0'` now.

oh, i used lv_port_pc_vscode with `[lvgl @ 8691574](https://github.com/lvgl/lvgl/tree/8691574c5bc8a84352d851d709e6d8c6d24fd624)`. the newer version of LVGL tell me `error: "Invalid drive letter"`,it`s very useful

closed 2025-02-05 by USTHzhanglu

vwheeler63, 2025-02-05:
@USTHzhanglu Hi!  In case it is helpful, this documentation was recently updated and is pending review/merging on exactly this topic.

http://crystal-clear-research.com/for_gabor/demo/docs.lvgl.io/master/details/libs/fs.html

Let me know if it is helpful.  (Some previous documentation did not spell out clearly how to use the file-system module correctly.)

USTHzhanglu, 2025-02-06:
> Hi! In case it is helpful, this documentation was recently updated and is pending review/merging on exactly this topic.

very helpful. I think some people are really easy to confuse something(like me)

vwheeler63, 2025-02-06:
> > Hi! In case it is helpful, this documentation was recently updated and is pending review/merging on exactly this topic.
> 
> very helpful. I think some people are really easy to confuse something(like me)

I wouldn't be hard on yourself:  it was confusing before!