lvgl/lvgl #7606: Docs Website Down?

vwheeler63, 2025-01-14:
### LVGL version

master

### Platform

Windows FireFox and Chrome Browsers

### What happened?

Attempting to access https://docs.lvgl.io/  from Denver, Colorado, USA right now (22:39 GMT, 14-Jan-2025) is producing an error:  NS_ERROR_UNKNOWN_HOST

FireFox:  Hmm. We’re having trouble finding that site.  We can’t connect to the server at docs.lvgl.io.
Chrome:  This site can’t be reached.  Check if there is a typo in docs.lvgl.io.

Command line:
```
> ping docs.lvgl.io
Ping request could not find host docs.lvgl.io. Please check the name and try again.
```

It appears the DNS 'A' record for it has disappeared.  https://dnschecker.org/#A/docs.lvgl.io   shows none of its base set of DNS servers have it.

[image]

dmytro-balytskyi, 2025-01-14:
Looks like it doesn't work either. 
I made a `git clone of https://github.com/lvgl/docs.git` to view the documentation locally.

kisvegabor, 2025-01-15:
Doesn't work here either. https://blog.lvgl.io is down too.

My CNAME record are in place:
[image]

I think the rest is done by GitHub as we are using GitHub pages for the docs. 

https://www.githubstatus.com/ doesn't report any issues.

Neothai, 2025-01-15:
I am having the same problem, can't access `https://docs.lvgl.io/` page on Chrome browser.
[image]

kisvegabor, 2025-01-15:
Thanks to @etag4048 we could resolve it. Actually it was incorrect for years. The CNAME records should point to `github.io` but they were pointing to `github.com`. 

It might take a few minutes while all the caches expire and it will work again on your side too.

vwheeler63, 2025-01-15:
> Thanks to @etag4048 we could resolve it.

Bravo!!

[image]

closed 2025-01-15 by vwheeler63