lvgl/lvgl #7614: LVGL Docs build failed

W-Mai, 2025-01-16:
### LVGL version

v9.2.0-388-g46ab31855

### Platform

CI

### What happened?

docs build break

W-Mai, 2025-01-16:
cc @kisvegabor

W-Mai, 2025-01-16:
should increase wasm initial mem or else?

[image]

vwheeler63, 2025-01-16:
Hi @W-Mai !   I'm not as familiar with the CI doc-build as I would like to be.  What is it that is being compiled?

vwheeler63, 2025-01-16:
Of note (and time coincident) is this change yesterday (about 5.5 hours ago) is:

[image]

AndreCostaaa, 2025-01-16:
there seems to be a recursive error somewhere that makes the building run out of memory

``` 
/home/runner/work/lvgl/lvgl/emscripten_builder/lv_conf.h:793:9: note: previous definition is here
#define LV_FS_DEFAULT_DRIVE_LETTER 'A'
        ^
1 warning generated.
[ 79%] Building C object lvgl/CMakeFiles/lvgl_examples.dir/examples/widgets/canvas/lv_example_canvas_5.c.o
In file included from /home/runner/work/lvgl/lvgl/emscripten_builder/lvgl/examples/widgets/canvas/lv_example_canvas_3.c:1:
In file included from /home/runner/work/lvgl/lvgl/emscripten_builder/lvgl/examples/widgets/canvas/../../lv_examples.h:16:
In file included from /home/runner/work/lvgl/lvgl/emscripten_builder/lvgl/src/../lvgl.h:130:
/home/runner/work/lvgl/lvgl/emscripten_builder/lvgl/src/lv_api_map_v9_1.h:84:9: warning: 'LV_FS_DEFAULT_DRIVE_LETTER' macro redefined [-Wmacro-redefined]
#define LV_FS_DEFAULT_DRIVE_LETTER     LV_FS_DEFAULT_DRIVER_LETTER
        ^
/home/runner/work/lvgl/lvgl/emscripten_builder/lv_conf.h:793:9: note: previous definition is here
#define LV_FS_DEFAULT_DRIVE_LETTER 'A'
        ^
1 warning generated.
``` 

This warning is generated recursively until the job runs out of memory and gets canceled

AndreCostaaa, 2025-01-16:
The line causing the crash was added on ad66f13f96 @vwheeler63. Not sure why it didn't cause problems before, something else must've changed but the first bad commit (124b2e268da6311804c60218cd502e4e70102478 @liamHowatt) doesn't modify anything about the docs generation nor the code #7620 should fix this

vwheeler63, 2025-01-16:
> [#7620](https://github.com/lvgl/lvgl/pull/7620) should fix this

Well done, André!!

liamHowatt, 2025-01-23:
https://github.com/lvgl/lv_web_emscripten/pull/26

The error was that Emscripten uses 16 MB by default as the initial memory for the program which wasn't enough to fit static variables. I increased it to 32 MB.

Andre's PR is important too.

kisvegabor, 2025-01-30:
https://github.com/lvgl/lv_web_emscripten/pull/26 is merged. Can we close this issue?

PR #7620 (2025-01-30): fix: warn user about deprecated LV_DEFAULT_DRIVE_LETTER
Fixes #7614
 

Like explained here: https://github.com/lvgl/lvgl/issues/7614#issuecomment-2596408267

The line `#define LV_FS_DEFAULT_DRIVE_LETTER     LV_FS_DEFAULT_DRIVER_LETTER` is causing the docs generation workflow to crash when building the examples because of a symbol redefinition. This checks if the symbol is already defined to avoid redefining it
closed 2025-01-30 by uLipe