lvgl/lvgl #7623: [Wayland] memory leak in _lv_wayland_flush

ThomasDevoogdt, 2025-01-17:
### LVGL version

master (v9.3.0-dev)

### Platform

Ubuntu 24.04.1 + Wayland, and also Buildroot 2024.08.1 + Weston.

### What happened?

I get this memory leak when doing something like `lv_wayland_window_create` & `lv_wayland_window_close`.

```
==547724== 96 bytes in 1 blocks are definitely lost in loss record 312 of 376
==547724==    at 0x484D953: calloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==547724==    by 0x48DA41C: ??? (in /usr/lib/x86_64-linux-gnu/libwayland-client.so.0.22.0)
==547724==    by 0x48DA5D4: wl_proxy_marshal_array_flags (in /usr/lib/x86_64-linux-gnu/libwayland-client.so.0.22.0)
==547724==    by 0x48DB407: wl_proxy_marshal_flags (in /usr/lib/x86_64-linux-gnu/libwayland-client.so.0.22.0)
==547724==    by 0x29369E: wl_surface_frame (wayland-client-protocol.h:3878)
==547724==    by 0x296411: _lv_wayland_flush (lv_wayland.c:2255)
==547724==    by 0x283176: call_flush_cb (lv_refr.c:1301)
==547724==    by 0x28301F: draw_buf_flush (lv_refr.c:1268)
==547724==    by 0x281981: refr_invalid_areas (lv_refr.c:621)
==547724==    by 0x28108C: lv_display_refr_timer (lv_refr.c:405)
==547724==    by 0x2B0CE9: lv_timer_exec (lv_timer.c:327)
==547724==    by 0x2B0320: lv_timer_handler (lv_timer.c:107)
```

I already tried some stuff to fix it, but couldn't find it easily. So will need some help.
This looks very similar to https://gitlab.freedesktop.org/wayland/weston/-/merge_requests/1537, but that fix is related to the composer.

I get another memory leak when calling `lv_wayland_window_create` again after the `lv_wayland_window_close`.

```
==549045== 16,690 (40 direct, 16,650 indirect) bytes in 1 blocks are definitely lost in loss record 385 of 387
==549045==    at 0x4846828: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==549045==    by 0x48E4898: wl_cursor_theme_load (in /usr/lib/x86_64-linux-gnu/libwayland-cursor.so.0.22.0)
==549045==    by 0x294EF8: handle_global (lv_wayland.c:1282)
==549045==    by 0x4FD2B15: ??? (in /usr/lib/x86_64-linux-gnu/libffi.so.8.1.4)
==549045==    by 0x4FCF3EE: ??? (in /usr/lib/x86_64-linux-gnu/libffi.so.8.1.4)
==549045==    by 0x4FD20BD: ffi_call (in /usr/lib/x86_64-linux-gnu/libffi.so.8.1.4)
==549045==    by 0x48D7BFD: ??? (in /usr/lib/x86_64-linux-gnu/libwayland-client.so.0.22.0)
==549045==    by 0x48D8472: ??? (in /usr/lib/x86_64-linux-gnu/libwayland-client.so.0.22.0)
==549045==    by 0x48D871B: wl_display_dispatch_queue_pending (in /usr/lib/x86_64-linux-gnu/libwayland-client.so.0.22.0)
==549045==    by 0x296B2E: wayland_init (lv_wayland.c:2450)
==549045==    by 0x296F71: lv_wayland_window_create (lv_wayland.c:2575)
```

Please, let me know if I have to compile/test something.

etag4048, 2025-01-23:
@ThomasDevoogdt I will try to investigate it this weekend, this driver needs some attention... And a refactor :)

kisvegabor, 2025-02-07:
@etag4048 are there any news on this topic?

closed 2025-03-09 by lvgl-bot