lvgl/lvgl #7639: Draw filled triangle (easy way)

focussing, 2025-01-21:
### LVGL version

v9.2.2

### Platform

Arduino using RP2040

### What happened?

I need to create a process diagram on my LCD and already used a circle and rectangle using quite simple functions:

```
void draw_rect(int i, int x, int y, int w, int h, lv_color_t bg_color, lv_color_t color) {
  rect[i] = lv_obj_create(lv_scr_act());
  lv_obj_set_scrollbar_mode(rect[i], LV_SCROLLBAR_MODE_OFF);
  lv_obj_set_size(rect[i], w, h);
  lv_obj_set_pos(rect[i], x, y);
  lv_obj_set_style_radius(rect[i], 0, 0);
  lv_obj_set_style_bg_color(rect[i], bg_color, 0);
  lv_obj_set_style_border_color(rect[i], color, 0);
  lv_obj_set_style_border_width(rect[i], 2, 0);
  // rect_color(i, color);
}

void draw_circle(int i, int x, int y) {
  circle[i] = lv_obj_create(lv_scr_act());
  lv_obj_set_size(circle[i], 40, 40);
  lv_obj_set_pos(circle[i], x, y);
  lv_obj_set_style_radius(circle[i], LV_RADIUS_CIRCLE, 0);
  lv_obj_set_style_border_color(circle[i], lv_color_black(), 0);
  lv_obj_set_style_border_width(circle[i], 2, 0);
  lv_obj_set_style_bg_color(circle[i], lv_color_white(), 0);
}

```
Now I have to draw a symbol of a valve and did it with the following simple code:

```
lv_obj_t *valve[2];

void draw_valve1(int i, int x, int y) {
  const int W = 30;
  const int H = 20;
  int X_L = x;
  int X_R = x + W;
  int Y_T = y;
  int Y_B = y + H;

  // create line
  static lv_point_precise_t line_points[] = { { X_L, Y_T }, { X_R, Y_B }, { X_R, Y_T }, { X_L, Y_B }, { X_L, Y_T } };
  valve[i] = lv_line_create(lv_screen_active());
  lv_line_set_points(valve[i], line_points, 5);
  lv_obj_add_style(valve[i], &style_line, 0);
}
```

Resulting in this:
[image]

Now I need to fill the triangles with color red to indicate it is closed, and green to indicate open. How can I achieve that?

etag4048, 2025-01-23:
LVGL can draw triangles,  [here](https://github.com/lvgl/lvgl/blob/master/src/draw/sw/lv_draw_sw_triangle.c ) However there is currently no widget for triangles.  You will need to setup a custom draw event ``LV_EVENT_DRAW_MAIN`` on the ``lv_obj_t`` a layer and draw unit.. Maybe someone on the forum might help you. It might be easier to use the canvas, LVGL can also draw vector graphics with ThorVG using the ``lv_canvas`` widget [here]( https://docs.lvgl.io/9.2/widgets/canvas.html)

AlexBestoso, 2025-01-29:
Also looking for a less confusing way to draw a triangle.
I need the ability to make the thing into a button; so I don't believe a canvas is an option…

`lv_draw_sw_triangle`  might work; but it's unclear how to init everything as there appears to be no official examples of how to set up draw units or how to draw triangles.

(which is odd to me seeing how graphics tends to build everything by using triangles)

etag4048, 2025-01-29:
@AlexBestoso Ok, I will write a quick guide on this.

liamHowatt, 2025-01-30:
To clarify, you do not create a draw unit or call `lv_draw_sw_triangle`  to draw triangles. You want to use `lv_draw_triangle_dsc_init` and `lv_draw_triangle` in a draw event callback.

Triangles are drawn in the widgets, music, ebike, and render demos in a draw event callback. A simpler docs example would be beneficial.

> which is odd to me seeing how graphics tends to build everything by using triangles

For 3D but not so much for 2D

focussing, 2025-01-30:
Hi @liamHowatt that looks promising!
Could you please provide a simple example of how do make it work?

Looking forward to your reaction :)

liamHowatt, 2025-02-03:
```c
static void my_obj_draw_end_cb(lv_event_t * e)
{
    lv_draw_triangle_dsc_t dsc;
    lv_draw_triangle_dsc_init(&dsc);
    dsc.p[0].x = 10;
    dsc.p[0].y = 10;
    dsc.p[1].x = 10;
    dsc.p[1].y = 50;
    dsc.p[2].x = 50;
    dsc.p[2].y = 50;
    dsc.bg_color = lv_palette_main(LV_PALETTE_GREEN);
    // dsc.bg_opa = ...;
    // dsc.bg_grad = ...;

    lv_obj_t * my_obj = lv_event_get_target_obj(e);
    lv_area_t my_obj_coords;
    lv_obj_get_coords(my_obj, &my_obj_coords);
    for(int i = 0; i < 3; i++) {
        dsc.p[i].x += my_obj_coords.x1;
        dsc.p[i].y += my_obj_coords.y1;
    }

    lv_layer_t * layer = lv_event_get_layer(e);
    lv_draw_triangle(layer, &dsc);
}

void triangle_simple(void)
{
    lv_obj_t * my_obj = lv_obj_create(lv_screen_active());
    lv_obj_center(my_obj);
    lv_obj_add_event_cb(my_obj, my_obj_draw_end_cb, LV_EVENT_DRAW_MAIN_END, NULL);
}
```

[image]

😃

AlexBestoso, 2025-02-08:
Howatts example worked exactly as advertised for me on an ESP32-s3 TFT_eSPI DMA configureation

focussing, 2025-02-21:
Hello @liamHowatt,

Your example works beautifully, thank you! In the callback I draw 2 triangles to create a valve symbol:

[image]

My question is how can I integrate this way of working into my system?
1) There is a square with rounded corners in which the valve is drawn, how do I suppress that?
2) The background color of a valve depends on whether it's open (green) or closed (red). That is determined in the process depending om several events/variables. So I need to have access to the symbol in order to set the color.
3) As your example enables me to set the background color, I can go back to black lines around the triangles. How can I draw outside lines?
4) In my example `draw_valve1` above there is no callback needed and also it does not use a layer, which solves 1) and 2). Is there a way to rewrite your example function to make it as simple as `draw_valve1`?

Looking forward to hearing from you.
Best regards,
Raymond

AlexBestoso, 2025-02-23:
You can use `lv_obj_set_style_*` to on the containing object to set the border opacity and color.

Which will make the background invisible; but it will still cover anything underneath the box. (problematic if you have buttons that get covered)

focussing, 2025-02-25:
@AlexBestoso that doesn't work, it sets the background color of the whole rectangle of the layer, not only the two triangles.

[image]

Any idea how to do that? I need to be able to switch the color of the 2 triangles whenever suitable. So basically the dsc.bg_color must be accessible from outside the callback (that's why I don't like the callback....

focussing, 2025-03-06:
Hi @liamHowatt 

Thanks a lot for your explanation! That helped a lot.
Especially your comment about `lv_obj_invalidate` was the right trigger.
I solved it now in te following way:

```
static void draw_valve_cb(lv_event_t *e) {
    Serial.print("CB ");

    int *uv = (int *)lv_event_get_user_data(e);
    Serial.println(*uv);

    int x = 5;
    int y = 5;

    const int W = 30;
    const int H = 20;

    int X_L = x;
    int X_R = x + W / 2;
    int Y_T = y;
    int Y_M = y + H / 2;
    int Y_B = y + H;

    lv_draw_triangle_dsc_t dsc;
    lv_draw_triangle_dsc_init(&dsc);
    dsc.p[0].x = X_L;
    dsc.p[0].y = Y_T;
    dsc.p[1].x = X_L;
    dsc.p[1].y = Y_B;
    dsc.p[2].x = X_R;
    dsc.p[2].y = Y_M;
    dsc.bg_color = valveColor;

    lv_obj_t *my_obj = lv_event_get_target_obj(e);
    lv_area_t my_obj_coords;
    lv_obj_get_coords(my_obj, &my_obj_coords);
    for (int i = 0; i < 3; i++) {
        dsc.p[i].x += my_obj_coords.x1;
        dsc.p[i].y += my_obj_coords.y1;
    }

    lv_layer_t *layer = lv_event_get_layer(e);
    lv_draw_triangle(layer, &dsc);

    X_L = x + W / 2;
    X_R = x + W;
    Y_T = y;
    Y_M = y + H / 2;
    Y_B = y + H;

    dsc.p[0].x = X_L;
    dsc.p[0].y = Y_M;
    dsc.p[1].x = X_R;
    dsc.p[1].y = Y_T;
    dsc.p[2].x = X_R;
    dsc.p[2].y = Y_B;

    for (int i = 0; i < 3; i++) {
        dsc.p[i].x += my_obj_coords.x1;
        dsc.p[i].y += my_obj_coords.y1;
    }

    lv_draw_triangle(layer, &dsc);
}

void draw_valve(int i) {
    Serial.print("draw valve ");
    Serial.println(i);

    vlv[i] = lv_obj_create(lv_screen_active());
    lv_obj_set_size(vlv[i], 40, 30);
    int x = 0, y = 0;
    switch (i) {
        case 0:
            x = 145;
            y = 35;
            break;
        case 1:
            x = 145;
            y = 155;
            break;
    }
    lv_obj_set_pos(vlv[i], x, y);
    lv_obj_set_style_border_width(vlv[i], 0, 0);
    lv_obj_set_style_bg_opa(vlv[i], LV_OPA_TRANSP, 0);

    static int user_data[] = { 1, 2 };
    lv_obj_add_event_cb(vlv[i], draw_valve_cb, LV_EVENT_DRAW_MAIN_END, &user_data[i]);
}

static void set_valve(int i, int open) {
    if (open) {
        valveColor = COLOR_OPEN;
        lv_obj_invalidate(vlv[i]);
    } else {
        valveColor = COLOR_CLOSED;
        lv_obj_invalidate(vlv[i]);
    }
}
```

closed 2025-03-24 by focussing

brightproject, 2025-06-26:
> static void my_obj_draw_end_cb(lv_event_t * e)

Hello @liamHowatt 

My terms - `LVGL 9.3` and `esp32s3`. 

I have a problem building the code.

```
error: 'struct lv_draw_triangle_dsc_t' has no member named 'bg_color'; did you mean 'color'?
  156 |     dsc.bg_color = lv_palette_main(LV_PALETTE_GREEN);
      |             ^~~~~~~~
      |             color
```

Line

> dsc.bg_color = lv_palette_main(LV_PALETTE_GREEN);

replaced with line

> dsc.color = lv_palette_main(LV_PALETTE_GREEN);

Please tell me, is it currently impossible to draw a filled triangle without a `canvas`?

As far as I understand, using a line you can draw an unfilled triangle, right?

liamHowatt, 2025-08-14:
Hi @brightproject,

Thanks, the API changed. I've edited my comments now.

> Please tell me, is it currently impossible to draw a filled triangle without a canvas?

Not impossible at all! This filled triangle is drawn without a `canvas`. https://github.com/lvgl/lvgl/issues/7639#issuecomment-2632197023

> As far as I understand, using a line you can draw an unfilled triangle, right?

Yes. That's currently the way to draw an unfilled triangle.