lvgl/lvgl #7640: How to create two rendering list of one lv_display?

yxGC, 2025-01-21:
### Introduce the problem

On low memory SoC system,  we can only afford 1/2 or 1/4  screen sized framebuffer.  And the less rendering buffer might cause more invoking of lv_obj drawn functions(when the framebuffer can't cover whole lv_obj). If we run codes on a NOR flash, the situation will be more worse.

On LVGL v9  we can fool LVGL that it has an whole framebuffer, so 'lv_display_refr_timer' will call all lv_objs drawn functions only once. Then we collect the drawn descriptions, and rendering them on anohter RTOS task which has hardware acclerating.

The porblems is:
How to collect next frame's drawn descriptions while current frame's drawn descriptions are still using?

### Proposal

Create two lv_display A&B,  and move all lv_objs to B lv_display while A lv_display is rendering?

yxGC, 2025-02-05:
@kisvegabor  Could you give me some adcices?

kisvegabor, 2025-02-25:
Sorry for the delay. Unfortunately I don't understand to root problem. So if LVGL render the screen in multiple steps (e.g. in case of 1/4 frame buffer), it's slow because of the the slow flash, right?

How will it be faster if LVGL give you all the draw tasks? If you had all the draw tasks,would you send them to a GPU, and run them e.g. 4 times using the screen areas (with 1/4 frame buffer)?

yxGC, 2025-02-26:
> Sorry for the delay. Unfortunately I don't understand to root problem. So if LVGL render the screen in multiple steps (e.g. in case of 1/4 frame buffer), it's slow because of the the slow flash, right?

Yes, it will invoke more codes. And if we have 1/8 frame buffer it will be more slower.

> How will it be faster if LVGL give you all the draw tasks? If you had all the draw tasks,would you send them to a GPU, and run them e.g. 4 times targeting different screen areas (with 1/4 frame buffer)?

GPU runing faster than LVGL, and it will invoke less codes(accessing less NOR flash).
If GPU have all the draw tasks and 1/4 frame buffer, it will draw 4 times automatically targeting to different areas.
And we exepect that LVGL can prepare all draw tasks of next frame, while GPU is rendering current frame.

kisvegabor, 2025-03-03:
Understood! To answer this:

> How to collect next frame's drawn descriptions while current frame's drawn descriptions are still using?

When LVGL collected all the draw tasks, you sent them to the GPU, the GPU is working on them, LVGL is waiting for rendering [here](https://github.com/lvgl/lvgl/blob/master/src/core/lv_refr.c#L1262) before starting the flushing. What you want, is handlning the GPU, and flushing in the background, and fool LVGL to start a new cycle immediately. 

It seems to me you need to trick LVGL on 2 places:
1. You can mark all the draw tasks as `LV_DRAW_TASK_STATE_READY` as you take them in `dispatch` so that the GPU won't wait for itself during rendring. (I assume that you are already doing that). It will unblock the loop [here](https://github.com/lvgl/lvgl/blob/master/src/core/lv_refr.c#L1262).  
2. You also need make your `flush_cb` return immediately, just manage the GPU and start sending the 1/4 frame size. Is it an SPI based display?.

The question is how will you tell LVGL before the real next frame that either GPU or the SPI is not ready. 
- For the GPU you can just `return 0` from the dispatcher when the GPU is not ready yet. LVGL wee keep trying to dispatch any new draw tasks. 
- For the `flush_cb` you should know that when a really frame is really transferred. If LVGL want's to send new data, just wait for completion.

yxGC, 2025-03-17:
Thanks for your reply, and it looks like it could work for me.
But I'm worried about the memories allocated in previous frame(lv_canvas buffer, for example, or some lv_draw_buf), they may be freed when LVGL start next cycle, right?

kisvegabor, 2025-03-24:
Ah, that's right. There some limitations in this regard and think even in principle it cannot be resolved. We can't save a reference to the buffers and keep using (potentially freeing or modifying) them in the meantime. 

I wonder if we can resolve the slow external flash issue differently as it's a generic problem. 

The obvious solution would be to move LVGL to RAM, but probably you don't have that much RAM though. We already have `LV_ATTRIBUTE_FAST_MEM` which is applied on some key functions, but I believe it can be used better and we can find more or different functions to move to RAM to improve performance.

yxGC, 2025-03-25:
What I want to do is another way:
1. Iterate through all the lv_objs within an invalidated area only once, regardless of the framebuffer's size, and gather all the drawing tasks.
2. Then, render parts of the invalidated area one by one in accordance with the framebuffer size.

This method can cut down the time spent on going through lv_objs. Iterating over lv_objs consumes a considerable amount of time when the framebuffer is extremely small. For instance, suppose we have a framebuffer that is 1/8 the size of the screen and an lv_img that covers the whole screen. When rendering this lv_img to the LCD, the draw event of this lv_img will be triggered 8 times, but only one of these times is actually necessary.

kisvegabor, 2025-04-10:
Sorry for being late again. I'd like to help, but I think it'd be better to think about it live in a meeting. Can you write me to gabor@lvgl.io?

yxGC, 2025-04-29:
> Can it be helpful? [#8142](https://github.com/lvgl/lvgl/pull/8142)

Yes, but not too much I think. It still create multi drawn tasks for one lv_obj if the

kisvegabor, 2025-04-29:
If you enable DEFERRED rendering, when dispatching is called for the first time all draw tasks will be already collected so you can do any magic with them.  I thought it might be helpful. 

However you still need to trick LVGL to think that you have a full screen buffer.

yxGC, 2025-04-30:
> If you enable DEFERRED rendering, when dispatching is called for the first time all draw tasks will be already collected so you can do any magic with them. I thought it might be helpful.
> 
> However you still need to trick LVGL to think that you have a full screen buffer.

OK, I got it. That will be helpful for me.  Thanks a lot.
By the way, did you have a plane to support deferred-rendering of multiple frames ?

kisvegabor, 2025-05-13:
> By the way, did you have a plane to support deferred-rendering of multiple frames ?

What do you mean by multiple fames?

yxGC, 2025-06-23:
But while the GPU is rendering the draw tasks of current frame, the CPU can prepare the draw tasks of next frame in parallel,  that is why I need  "draw tasks of multiple frames ".
But you said there is an buffer limitation that can't be resolved at [here](https://github.com/lvgl/lvgl/issues/7640#issuecomment-2749481038).
So I want to know if you have a plane to  resolve the limitation.

closed 2025-07-26 by lvgl-bot