lvgl/lvgl #7656: lv_disp_get_inactive_time resets without touch every ~500 ms

dontbanmeplz, 2025-01-24:
### LVGL version

v8.4.0

### Platform

https://lilygo.cc/products/t-watch-s3-us

### What happened?

When calling lv_disp_get_inactive_time(NULL)) consistently my sleep function works fine, but as soon as I stop calling it the timer seems to reset by itself every ~500ms for a reason i cant figure out.

I stop calling it by toggling the bool `disableSleep`

I expected it to just count up and reset when i touch the screen

### How to reproduce?

```cpp
void sleepHandler() {
	void enterLightSleep();
	if (!disableSleep)
		Serial.printf("Inactive time: %d\n", lv_disp_get_inactive_time(NULL));
		if (lv_disp_get_inactive_time(NULL) >= screenTimeout)
			enterLightSleep();
}

void loop()
{
	lv_task_handler();
	sleepHandler();
...
}
```
short output snippet:
```
Inactive time: 450
Inactive time: 456
Inactive time: 464
Inactive time: 470
Inactive time: 476
Inactive time: 482
Inactive time: 488
Inactive time: 496
Inactive time: 502
Inactive time: 375
Inactive time: 381
Inactive time: 389
Inactive time: 395
Inactive time: 401
Inactive time: 407
Inactive time: 413
Inactive time: 421
Inactive time: 427
Inactive time: 433
Inactive time: 439
Inactive time: 445
Inactive time: 453
Inactive time: 459
Inactive time: 465
Inactive time: 471
Inactive time: 477
Inactive time: 485
Inactive time: 491
Inactive time: 497
Inactive time: 503
Inactive time: 66
Inactive time: 73
Inactive time: 81
Inactive time: 87
Inactive time: 93
Inactive time: 99
Inactive time: 105
Inactive time: 113
Inactive time: 119
Inactive time: 125
Inactive time: 131
Inactive time: 137
Inactive time: 145
Inactive time: 151
Inactive time: 157
Inactive time: 163
Inactive time: 169
Inactive time: 177
Inactive time: 183
Inactive time: 189
Inactive time: 195
Inactive time: 201
Inactive time: 209
Inactive time: 215
Inactive time: 221
Inactive time: 227
Inactive time: 233
Inactive time: 241
Inactive time: 247
Inactive time: 253
Inactive time: 259
Inactive time: 265
Inactive time: 273
Inactive time: 279
Inactive time: 285
Inactive time: 291
Inactive time: 297
Inactive time: 305
Inactive time: 311
Inactive time: 317
Inactive time: 323
Inactive time: 329
Inactive time: 337
Inactive time: 343
Inactive time: 349
Inactive time: 355
Inactive time: 361
Inactive time: 369
Inactive time: 375
Inactive time: 381
Inactive time: 387
Inactive time: 393
Inactive time: 401
Inactive time: 407
Inactive time: 413
Inactive time: 419
Inactive time: 425
Inactive time: 433
Inactive time: 439
Inactive time: 445
Inactive time: 451
Inactive time: 457
Inactive time: 465
Inactive time: 471
Inactive time: 477
Inactive time: 483
Inactive time: 489
Inactive time: 497
Inactive time: 503
Inactive time: 66
Inactive time: 72
Inactive time: 80
Inactive time: 86
Inactive time: 92
Inactive time: 98
Inactive time: 104
Inactive time: 112
Inactive time: 118
Inactive time: 124
Inactive time: 130
Inactive time: 136
Inactive time: 144
Inactive time: 150
Inactive time: 156
Inactive time: 162
Inactive time: 168
Inactive time: 176
Inactive time: 182
Inactive time: 188
Inactive time: 194
Inactive time: 200
Inactive time: 208
Inactive time: 214
Inactive time: 220
Inactive time: 226
Inactive time: 232
Inactive time: 240
Inactive time: 246
Inactive time: 252
Inactive time: 258
Inactive time: 264
Inactive time: 272
Inactive time: 278
Inactive time: 284
Inactive time: 290
Inactive time: 296
Inactive time: 304
Inactive time: 310
Inactive time: 316
Inactive time: 322
Inactive time: 328
Inactive time: 336
Inactive time: 342
Inactive time: 348
Inactive time: 354
Inactive time: 360
Inactive time: 368
Inactive time: 374
Inactive time: 380
Inactive time: 386
Inactive time: 392
Inactive time: 400
Inactive time: 406
Inactive time: 412
Inactive time: 418
Inactive time: 424
Inactive time: 432
Inactive time: 438
Inactive time: 444
Inactive time: 450
Inactive time: 456
Inactive time: 464
Inactive time: 470
Inactive time: 476
Inactive time: 482
Inactive time: 488
Inactive time: 496
Inactive time: 502
Inactive time: 66
Inactive time: 72
Inactive time: 80
Inactive time: 86
```

dontbanmeplz, 2025-01-27:
I just wrote a method in my touch capture function to handle this. The lvgl one just doesnt work

closed 2025-01-27 by dontbanmeplz

kisvegabor, 2025-01-30:
I have no idea what could trigger it in every 500 ms. 🙁