lvgl/lvgl #7660: layer stride not update after lv_display_set_rotation

Villaden74, 2025-01-26:
### LVGL version

v9.2.2

### Platform

Software: Espressif-IDE, esp-idf 5.3.1, LVGL9.2.2, esp_lvgl_port 2.4.3
Hardware: esp32-s3 with display touch 320 (coloumns) x 480 (rows) (AXS15231b) bought on AliExpress, display is QSPI driven.

### What happened?

Even if it is documented in datasheet, the low level driver command CASET hasn't effect so I have to set software rotation and full refresh.
I also need to rotate it 90° (landscape).
But, after set the rotation via lv_display_set_rotation, the display shows a corrupted version of framebuffer with 1/2 aliases side-by-side.

I can resolve it changing the stride parameter of layer_reshape_draw_buf function iinside refr_area function placed in /src/core/lv_refr.c.
I changed it from layer->draw_buf->header.stride to LV_STRIDE_AUTO.
It seems that after rotation the  layer->draw_buf->header.stride from the initially 640 doesn't change to 960.
What do you think about it?

FAksel, 2025-01-28:
Hello, I have the same issue (esp-idf v5.4, LVGL 9.3.0dev, using the esp_lcd component.
I am writting an esp_lcd driver for the sharp memory display 2.7 (spi monochrome, 400x240).

My driver works for rotation = 0, (I use LV_COLOR_FORMAT_I1, LV_DISPLAY_RENDER_MODE_FULL and a full display buffer of size (240*400 / 8 + 8).

If I call lv_display_set_rotation to 90, the lv_timer_handler hangs and I get a watchdog timeout.
The problem and solution is the same as above. the buffer header stride is not updated and lv_draw_buf_reshape returns buf too small for new shape.

Changing `void refr_area(const lv_area_t * area_p)` in lv_refr.c
from
```
if (disp_refr->render_mode == LV_DISPLAY_RENDER_MODE_FULL) {
        /*In full mode the area is always the full screen, so the buffer area to it too*/
        layer->buf_area = *area_p;
        layer_reshape_draw_buf(layer, layer->draw_buf->header.stride);
}
```
to
```
if (disp_refr->render_mode == LV_DISPLAY_RENDER_MODE_FULL) {
        /*In full mode the area is always the full screen, so the buffer area to it too*/
        layer->buf_area = *area_p;
        layer_reshape_draw_buf(layer, LV_STRIDE_AUTO);
}
```

solves the issue

liamHowatt, 2025-01-30:
Thanks @FAksel, please open a PR for it if you want

FAksel, 2025-02-21:
Issue #7797 is the same. Commenting so it doesnt close.
 
@liamHowatt This is a drity fix, The root cause needs to be addressed: the buffer stride is not updated when lv_display_set_rotation is called.

Secondary question: what is the reasonning behind stalling and closing issues automatically? It is totally ok if it takes time, but bugs need to be addressed at some point no?

PR #8107 (2025-05-11): feat(disp): allow rotation with `FULL` render mode
Fixes #7660
Fixes #7797

See the issues for more context.

Test in SDL with this:
https://github.com/liamHowatt/lvgl/commit/8f4241abb044cb047a48b4892672a8b15af15db9
I am not committing the SDL changes because it would make `FULL` slower for the non-rotated case.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new o [... cut]
closed 2025-05-11 by FASTSHIFT