lvgl/lvgl #7661: lv_os.h not included unless LV_USE_PRIVATE_API

faxe1008, 2025-01-26:
### LVGL version

4744bbe194b492fdbeaeecaf19a7c4a710984a12

### Platform

Zephyr integration

### What happened?

Build error with:

```
error: unknown type name ‘lv_thread_prio_t’; did you mean ‘lv_thread_sync_t’?
   16 | lv_result_t lv_thread_init(lv_thread_t *thread, const char *const name, lv_thread_prio_t prio,
      |                                                                         ^~~~~~~~~~~~~~~~
      |                                                                         lv_thread_sync_t
```
The type `lv_thread_prio_t` is only visible to out-of tree implementers if `LV_USE_PRIVATE_API` is set. This should imo not be the case since it's kind of a public interface.

kisvegabor, 2025-01-28:
I have intentionally made LVGL OS API private, because IMO it should be an internal LVGL thing. 

In a larger application where you have many libraries (e.g. LVGL, WiFi, Bluetooth, Audio, etc) why would you rely on the OS abstraction of a particular library?

So LVGL is a library and not framework. Therefore it shouldn't be responsible for OS abstraction for the whole app. Instead you should use the OS as it is to create a UI thread if needed.

This is only my current understanding, and I'm very curious of your opinion.

kisvegabor, 2025-01-29:
For this we have [`LV_OS_CUSTOM`](https://github.com/lvgl/lvgl/blob/master/src/osal/lv_os.h#L40-L42). So you just need to specify a header file for the custom OSAL and implement the requires functions somewhere. 

In such an advanced use-case I don't think it's an issue of the developers need to include some private headers.

faxe1008, 2025-02-06:
Yeah, alright. I guess it's kind of a taste preference of when its considered to be "okay" to use private headers.

closed 2025-02-06 by faxe1008