lvgl/lvgl #7663: gif does not work properly

zhangweiquan815, 2025-01-27:
### LVGL version

v9.2.2

### Platform

project : lv_port_pc_vscode
lvgl version : v9.2.2
platform ：win11
IDE : vscode
compiler ： ming64w

### What happened?

Use GIF as a routine. Only the first frame of the GIF is displayed. log without exception, I traced the code and found that it seemed to repeat decoding and displaying the first frame, how can I avoid this problem.

### How to reproduce?

  lv_obj_t * gif = lv_gif_create(lv_layer_top());
  lv_gif_set_src(gif,"D:bulb.gif");
  lv_obj_align(gif, LV_ALIGN_RIGHT_MID, 0, 0);

liamHowatt, 2025-01-28:
I could not reproduce the issue. Please try out #7675 which fixes another GIF issue, to see if that fixes it.

zhangweiquan815, 2025-02-05:
I used the gifdec.c code for the master branch. Problems can still arise. The tag for the pull code is v9.2.2

zhangweiquan815, 2025-02-05:
I turn on the debugging function. define LV_USE_REFR_DEBUG 1. I can see the refresh area of the gif flashing, but the image area only shows the first piece of content. The following content is not refreshed.

zhangweiquan815, 2025-02-05:
> I could not reproduce the issue. Please try out [#7675](https://github.com/lvgl/lvgl/pull/7675) which fixes another GIF issue, to see if that fixes it.

I'm trying to track the gif parsing process. The process was so complicated that I didn't find the problem. I guess the gif refresh process is normal. Maybe it's stuck somewhere parsing frames.

zhangweiquan815, 2025-02-05:
`/**
 * @file lv_conf.h
 * Configuration file for v9.2.2
 */

/*
 * Copy this file as `lv_conf.h`
 * 1. simply next to the `lvgl` folder
 * 2. or any other places and
 *    - define `LV_CONF_INCLUDE_SIMPLE`
 *    - add the path as include path
 */

/* clang-format off */
#if 1 /*Set it to "1" to enable content*/

#ifndef LV_CONF_H
#define LV_CONF_H

/*If you need to include anything here, do it inside the `__ASSEMBLY__` guard */
#if  0 && defined(__ASSEMBLY__)
#include "my_include.h"
#endif

/*====================
   COLOR SETTINGS
 *====================*/

/*Color depth: 1 (I1), 8 (L8), 16 (RGB565), 24 (RGB888), 32 (XRGB8888)*/
#define LV_COLOR_DEPTH 32

/*=========================
   STDLIB WRAPPER SETTINGS
 *=========================*/

/* Possible values
 * - LV_STDLIB_BUILTIN:     LVGL's built in implementation
 * - LV_STDLIB_CLIB:        Standard C functions, like malloc, strlen, etc
 * - LV_STDLIB_MICROPYTHON: MicroPython implementation
 * - LV_STDLIB_RTTHREAD:    RT-Thread implementation
 * - LV_STDLIB_CUSTOM:      Implement the functions externally
 */
#define LV_USE_STDLIB_MALLOC    LV_STDLIB_BUILTIN
#define LV_USE_STDLIB_STRING    LV_STDLIB_BUILTIN
#define LV_USE_STDLIB_SPRINTF   LV_STDLIB_BUILTIN

#define LV_STDINT_INCLUDE       <stdint.h>
#define LV_STDDEF_INCLUDE       <stddef.h>
#define LV_STDBOOL_INCLUDE      <stdbool.h>
#define LV_INTTYPES_INCLUDE     <inttypes.h>
#define LV_LIMITS_INCLUDE       <limits.h>
#define LV_STDARG_INCLUDE       <stdarg.h>

#if LV_USE_STDLIB_MALLOC == LV_STDLIB_BUILTIN
    /*Size of the memory available for `lv_malloc()` in bytes (>= 2kB)*/
    #define LV_MEM_SIZE (8 * 1024 * 1024U)          /*[bytes]*/

    /*Size of the memory expand for `lv_malloc()` in bytes*/
    #define LV_MEM_POOL_EXPAND_SIZE 0

    /*Set an address for the memory pool instead of allocating it as a normal array. Can be in external SRAM too.*/
    #define LV_MEM_ADR 0     /*0: unused*/
    /*Instead of an address give a memory allocator that will be called to get a memory pool for LVGL. E.g. my_malloc*/
    #if LV_MEM_ADR == 0
        #undef LV_MEM_POOL_INCLUDE
        #undef LV_MEM_POOL_ALLOC
    #endif
#endif  /*LV_USE_STDLIB_MALLOC == LV_STDLIB_BUILTIN*/

/*====================
[... 1033 lines cut ...]
    #define LV_DEMO_MUSIC_LARGE     0
    #define LV_DEMO_MUSIC_AUTO_PLAY 0
#endif

/*Flex layout demo*/
#define LV_USE_DEMO_FLEX_LAYOUT     0

/*Smart-phone like multi-language demo*/
#define LV_USE_DEMO_MULTILANG       0

/*Widget transformation demo*/
#define LV_USE_DEMO_TRANSFORM       0

/*Demonstrate scroll settings*/
#define LV_USE_DEMO_SCROLL          0

/*Vector graphic demo*/
#define LV_USE_DEMO_VECTOR_GRAPHIC  0

/*--END OF LV_CONF_H--*/

#endif /*LV_CONF_H*/

#endif /*End of "Content enable"*/
`

zhangweiquan815, 2025-02-05:
This is my configuration of lvgl. I hope you can give me some suggestions, such as updating the version or integrating the patch, thank you.

zhangweiquan815, 2025-02-08:
I found an issue with gif not working. The lv_port_pc_vscode project is not very compatible with SDL and lvgl. demo works and displays normally, but gif and Canvas always show only the first frame of the picture. No matter how I refresh it later. I made sure that each frame of the gif image was parsed correctly and the Canvas was drawn multiple times.

zhangweiquan815, 2025-02-08:
I found an issue with gif not working. The lv_port_pc_vscode project is not very compatible with SDL and lvgl. demo works and displays normally, but gif and Canvas always show only the first frame of the picture. No matter how I refresh it later. I made sure that each frame of the gif image was parsed correctly and the Canvas was drawn multiple times. Maybe I should go to the lv_port_pc_vscode repository and submit this problem. I will close this issue.

closed 2025-02-08 by zhangweiquan815