lvgl/lvgl #7672: Whole screen gets invalidated

jeanthom, 2025-01-28:
### LVGL version

v9.2.0-457-g0f67eb67c

### Platform

RP2040 on bare metal

### What happened?

I have a widget occupying the whole screen (external SPI display, DMA driven). Whenever the widget is modified, a small region of it is invalidated. However when testing in real life situation with LV_USE_REFR_DEBUG=1, I observe that the whole screen gets a redraw even if the region isn't affected.

jeanthom, 2025-01-28:
Digging through the pending pull requests, my issue might be related to https://github.com/lvgl/lvgl/pull/7598/files. Will try this patch tomorrow.

jeanthom, 2025-01-29:
I just tested #7598, and it fixed my issue.

liamHowatt, 2025-01-30:
Nice!

closed 2025-01-30 by liamHowatt