lvgl/lvgl #7686: Redundant Document?

vwheeler63, 2025-01-30:
### Introduce the problem

https://docs.lvgl.io/master/details/libs/fs.html

seems redundant with

https://docs.lvgl.io/master/details/main-components/fs.html

the latter of which appears is a much more thorough coverage of the same topic.  I see nothing in the former that is not already covered in the latter.

### Proposal

Eliminate https://docs.lvgl.io/master/details/libs/fs.html

kisvegabor, 2025-02-01:
`libs/fs` should be just an overview to list the possible 3rd library interfaces. 

`main-components/fs` describes how to create new drivers. 

> Eliminate https://docs.lvgl.io/master/details/libs/fs.html

IMO we should mention somewhere an simple list of possible drivers. (It can be at the beginning of the `main-components` docs). Maybe can we link a single docs in both `libs` and `main-components` ?

vwheeler63, 2025-02-02:
Hmmmm....  (`*realization*`)  In writing my answer to you, I realized that `main-components/fs.rst` was written to be a core component of LVGL, whereas the list of drivers presented in the introduction of `libs/fs.rst` were all written by 3rd parties and contributed to LVGL.  (I was previously interpreting those drivers as being part of that core component.)

Is that correct?

If so, then my *real* objection to `libs/fs.rst` is merely that there is duplication of material.

If that is correct, then [this section](http://crystal-clear-research.com/for_gabor/demo/docs.lvgl.io/master/details/main-components/fs.html#drivers-that-come-with-lvgl) should probably be presented to present in a way so as to carry the concept that those are 3rd-party-contributed file-system drivers.

Does that align with the actual relationship between `main-components/fs.rst` and the list of drivers?

If so, then all I need to is re-work that lower part of `libs/fs.rst` so as to refer to sections of `main-component/fs.rst` for more details (because it is better explained how to use the `..._DRIVER_LETTER` value).

vwheeler63, 2025-02-04:
@kisvegabor  With the realization above, the solution is a no-brainer.  I know what to do.  :-)  (Closing this since the solution is so obvious.)

kisvegabor, 2025-02-05:
> If so, then all I need to is re-work that lower part of libs/fs.rst so as to refer to sections of main-component/fs.rst for more details (because it is better explained how to use the ..._DRIVER_LETTER value).

Confirmed! 😊