lvgl/lvgl #7687: File System Driver Letters

vwheeler63, 2025-01-30:
### Introduce the problem

cc:  @liamHowatt , @kisvegabor 

Hi, Liam!  Technical question:  in `lv_conf.h`, just below `LV_FS_DEFAULT_DRIVER_LETTER` is a whole set of file system macros, each set having an `LV_FS_..._LETTER` which, at a glance, APPEARS to all be used in the same way `LV_FS_WIN32_LETTER` is -- as a driver identifier -- not a DRIVE LETTER.  Looking at `lv_fs_littlefs.c`, it indeed gets passed to the `lv_fs_get_drv()` function, which uses it that way.  On closer examination, not all of them get passed to the `lv_fs_get_drv()` function, but they do all seem to either get assigned to `lv_fs_drv_t::letter` or otherwise get used this way through locally-defined functions in that particular driver's source code.  You seem to have more experience with this part of LVGL.  Can you confirm this is indeed what they all are?

The reason I ask is that 2 things are troubling me right now because of some deceptive remnants of the concept of "DRIVE LETTER" still remaining in code (which is going to 100% confuse Windows, DOS, and OS/2 users):

- `/lvgl/lv_conf_template.h` is still filled with deceptive comments next to these macros (which comments are supposed to provide instruction to the end user as to what they mean, and thus how to use them.  See example below.
- A number of function arguments still use the name "drive_letter" which is itself deceptive, and is simultaneously a documentation issue due to the role the argument name has in helping the LVGL user/programmer understand how to use these values.

The example that brought my attention to this is:
```c
    #define LV_FS_LITTLEFS_LETTER '\0'  /**< Set an upper cased letter on which the drive will accessible (e.g. 'A') */
```

which is going to lead many programmers astray.  Other examples:
```c
static void lv_fs_drv_uefi_init(lv_fs_drv_t * drv, char fs_drive_letter, EFI_HANDLE fs_handle) { ... }

void lv_fs_make_path_from_buffer(lv_fs_path_ex_t * path, char letter, const void * buf, uint32_t size) { ... }

lv_fs_drv_t * lv_fs_get_drv(char letter) { ... }
```

which is probably not a complete list, but is a quick list of functions where these macros are being passed.

My think is -- since we clarified the meaning of this for `LV_FS_DEFAULT_DRIVER_LETTER` -- that I should go ahead and fix the comments next to the remaining `LV_FS_..._LETTER` macros, and argument names where they are used -- along with the API documentation in the accompanying comments.  If my understanding of these are correct, the correct name for this concept is (as before):  "driver identifier" or "driver id".

Just wanting to confirm before I leap into changing code.  I think it will help all programmers for whom a "drive letter" is the leading part of any file system full path.

### Proposal

- Correct/clarify comments next to (regex) /#define LV_FS_.*_LETTER\s+/ macros in `lv_conf_template.h`.
- Regenerate `lv_conf_template.h` from the changed `lv_conf_template.h`.
- Find all functions where these macros are used and review/clarify:
  - associated argument names, and
  - API documentation that documents them.

My guess:  30 minutes to complete.

Kind regards,
Vic

liamHowatt, 2025-02-04:
I can confirm that any config in called `LV_FS_*_LETTER` refers to a driver letter that will be used to access files through that driver when using `lv_fs` file functions.

vwheeler63, 2025-02-04:
@kisvegabor Hi, Gábor!  Would you condone by proceeding with the above?

PR #7710 (2025-02-12): feat(fs):  clarify rest of docs on driver-identifier letters.
A previous fix to this in PR #7440 overlooked a handful of documentation that still carried the misleading concepts.  That oversight is handled in this PR.

Resolves #7687

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.  Done.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.  n/a
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.  n/a
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](ht [... cut]
closed 2025-02-12 by liamHowatt