lvgl/lvgl #7698: ESP32-P4 LVGL9.2.2 Display tearing in certain specific scenarios

SSEHX, 2025-01-31:
### LVGL version

V9.2.2

### Platform

ESP32P4 ESP-IDF 5.4

### What happened?

ESP32 When ESP32P4 has no background elements, the floating FPS indicator will display tearing, including other elements. However, when hovering on the Lottie animation, tearing will not occur and everything is normal,

https://github.com/user-attachments/assets/c1cc8c5e-2c33-4cb5-a707-5a0213a977cc

[image]

[image]

### How to reproduce?

lv_obj_t* create_solid_background(lv_color_t color, lv_obj_t* parent) {
    // 若未指定父对象，则使用当前活动屏幕
    if (!parent) parent = lv_scr_act();
    
    // 创建基础对象
    lv_obj_t *bg = lv_obj_create(parent);
    
    // 设置全屏尺寸
    lv_obj_set_size(bg, LV_PCT(100), LV_PCT(100));
    
    // 设置样式
    lv_obj_set_style_bg_color(bg, color, LV_PART_MAIN);
    lv_obj_set_style_border_width(bg, 0, LV_PART_MAIN);
    lv_obj_set_style_radius(bg, 0, LV_PART_MAIN);
    lv_obj_set_style_pad_all(bg, 0, LV_PART_MAIN);
    
    // 确保置于底层
    if (lv_obj_get_parent(bg) == parent) {
        lv_obj_move_to_index(bg, 0); // LVGL v8.3+
        // 或 lv_obj_move_background(bg); // 兼容旧版本
    }
    
    return bg;
}

SSEHX, 2025-01-31:
screen :410x502
buffer: single 410x55
qspi dma

closed 2025-02-26 by lvgl-bot

liamHowatt, 2025-03-18:
It looks like the palette is not being skipped for the QR code. Could be be related to https://github.com/lvgl/lvgl/issues/6965#issuecomment-2423542724 ? Can you ensure `LV_DRAW_SW_SUPPORT_ARGB8888` is enabled?