lvgl/lvgl #7701: Bug in `lv_obj_remove_event_cb_with_user_data()`?

vwheeler63, 2025-02-02:
### LVGL version

v9.3

### Platform

While re-examining memory allocation / clean-up operations with Observer for compatibility (since I am involved in writing explicit instructions for programmers using LVGL to correctly use Observer), I noticed something that appears to be a bug:

1.  There is nothing that prevents `lv_subject_deinit()` from being run on a subject that has had Widget properties bound to it with the `..._bind_...()` functions.  And indeed, it looks like it was intended to handle this case.
2. In the middle of that function's `while(observer)` loop (looping through linked-list of Observers):

```c
        if(observer->for_obj) {
            lv_obj_remove_event_cb(observer->target, unsubscribe_on_delete_cb);
            lv_obj_remove_event_cb_with_user_data(observer->target, NULL, subject);
        }
```

correctly removes the Widget's `LV_EVENT_DELETE` event with `lv_obj_remove_event_cb()` (which will be present in all cases where `observer->for_obj == 1`).  However, the next line APPEARS to be for the purpose of deleting `LV_EVENT_VALUE_CHANGED` events for Widgets where an Observer is implementing a binding to the Widget's value.  The call to `lv_obj_remove_event_cb_with_user_data()` is being made with a `NULL` callback argument, which, from what I can see, will NEVER remove anything, because removal of an event only happens `if(event_dsc->cb == NULL)`, which I have never seen any case where an event would have a NULL call-back function pointer associated with an event.  And certainly Observers that implement bound Widget Value Properties CERTAINLY DO have valid (non-NULL) callback pointers (each Widget type has its own callback for this).  If I am understanding the intention of that 2nd line correctly, it will not work and will leave the `subject` attached to the Widget's `LV_EVENT_VALUE_CHANGED` event.

If `subject` is a global variable, while technically this "doesn't hurt anything", it will leave the Widget changing the `subject`'s value each time the user interacts with it and changes the Widget's value, instead of "completely disconnecting" `subject` from everything in the system as it appears was its design intention for the `lv_subject_deinit()` function.

If `subject` was dynamically allocated, and if `subject` subsequently was deleted before the Widget was deleted, leaving that event connected (with a now-invalid pointer to `subject` in the event's `user_data`) could be catastrophic and cause a system crash due to accessing an invalid pointer.

If `lv_obj_remove_event_cb_with_user_data()` was meant to accept a NULL argument as a "wildcard" (i.e. to accept any callback) then:

- `lv_obj_remove_event_cb_with_user_data()` the IF condition inside the FOR loop never implemented it that way (perhaps an overlooked step when `lv_subject_deinit()` was written?).

### What happened?

Bug observed by documenting object hierarchy for the different Subject-Observer relationship types to double-verify my clean-up instructions in `observer.rst` were correct.

### How to reproduce?

See above.

### Propose:

Assuming `lv_obj_remove_event_cb_with_user_data()` was meant to accept NULL as a wildcard (matching every callback):

#### Step 1

`lv_obj_event.c`:164 be changed from
```c
        if(dsc && dsc->cb == event_cb && dsc->user_data == user_data) {
```
to
```c
        if(dsc && (event_cb == NULL || dsc->cb == event_cb) && dsc->user_data == user_data) {
```

Note:  `lv_subject_deinit()` is the only call in the whole LVGL library that passes NULL for this argument.

#### Step 2

API documentation for `lv_obj_remove_event_cb_with_user_data()` updated to indicate NULL can be passed for the `event_cb` argument to match all events.

---

If `lv_obj_remove_event_cb_with_user_data()` was NOT meant to accept NULL as a wildcard (matching every callback):

#### Step 1

`lv_subject_deinit()` have the call to `lv_obj_remove_event_cb_with_user_data()` replaced with 4 calls:

```c
            lv_obj_remove_event_cb(observer->target, arc_value_changed_event_cb);
            lv_obj_remove_event_cb(observer->target, slider_value_changed_event_cb);
            lv_obj_remove_event_cb(observer->target, roller_value_changed_event_cb);
            lv_obj_remove_event_cb(observer->target, dropdown_value_changed_event_cb);
```

Although inefficient, it's the only way to make it correct that I can see.

vwheeler63, 2025-02-03:
cc:  @kisvegabor Gábor, can you please have a look at this?  You are the original implementer of the loop inside `lv_obj_remove_event_cb_with_user_data()` and if NULL was meant to be a wildcard for the callback, this is an easy fix.

kisvegabor, 2025-02-04:
Hi Vic,

It was really an oversight and `NULL` should be used as a wildcard. 

>    if(dsc && (event_cb == NULL || dsc->cb == event_cb) && dsc->user_data == user_data) {

Please update `lv_obj_remove_event_cb_with_user_data` to this.

vwheeler63, 2025-02-04:
> Please update `lv_obj_remove_event_cb_with_user_data` to this.

Done!

PR #7651 (2025-02-11): Don't squash - proofread/edit docs batch 16
This is a fairly thorough re-working of the Observer documentation to clarify how it is used and to make it easier to learn.  It includes its API documentation.

- observer.rst  (overhaul)  [[link](http://crystal-clear-research.com/for_gabor/demo/docs.lvgl.io/master/details/other-components/observer.html)]  <-- how generated document looks.
- lv_observer_private.h  (API documentation)
- lv_observer.h
  - API documentation,
  - 1 arg name to clarify meaning.
- lv_observer.c
  - 1 arg name to clarify meaning,
  - capitalization and grammar on 2 interna [... cut]
commit 0ae888b6 (2025-02-11): fix(event): consider NULL as a wildcard for callback arg in lv_obj_remove_event_cb_with_user_data()

Resolves #7701
closed 2025-02-11 by kisvegabor