lvgl/lvgl #7702: Make `lv_obj_remove_from_subject()` legal for Widget Binding?

vwheeler63, 2025-02-02:
### Introduce the problem

Currently `lv_obj_remove_from_subject()` cannot be called when a Widget's `value` property has been bound to a Subject, because it leaves the Widget's `LV_EVENT_VALUE_CHANGED` event still attached to the subject.  I observed while writing "clean-up" instructions in `observer.rst` that keeping track of when NOT to use `lv_obj_remove_from_subject()` seems unnecessarily complicated because of this.  And since `lv_subject_deinit()` gracefully picks up this case, it seems like `lv_obj_remove_from_subject()` should also.

### Proposal

Assuming issue #7701 is a correctly-observed bug/solution:

If we want `lv_obj_remove_from_subject()` to pick up the case of gracefully handling when a Widget's `value` property has been bound to a Subject, it will need to delete the Widget's `LV_EVENT_VALUE_CHANGED` event by making this call once *outside the FOR loop*:

```c
lv_obj_remove_event_cb_with_user_data(obj, NULL, subject);
```

kisvegabor, 2025-02-04:
See my comment in [#7701](https://github.com/lvgl/lvgl/issues/7701)

vwheeler63, 2025-02-04:
> See my comment in [#7701](https://github.com/lvgl/lvgl/issues/7701)

Thank you!

PR #7651 (2025-02-11): Don't squash - proofread/edit docs batch 16
This is a fairly thorough re-working of the Observer documentation to clarify how it is used and to make it easier to learn.  It includes its API documentation.

- observer.rst  (overhaul)  [[link](http://crystal-clear-research.com/for_gabor/demo/docs.lvgl.io/master/details/other-components/observer.html)]  <-- how generated document looks.
- lv_observer_private.h  (API documentation)
- lv_observer.h
  - API documentation,
  - 1 arg name to clarify meaning.
- lv_observer.c
  - 1 arg name to clarify meaning,
  - capitalization and grammar on 2 interna [... cut]
commit 9b3c365a (2025-02-11): feat(observer):  Make `lv_obj_remove_from_subject()` legal for Widget Binding.

Resolves #7702
closed 2025-02-11 by kisvegabor