lvgl/lvgl #7703: NEMA_GFX does not build

tdjastrzebski, 2025-02-03:
### LVGL version

9.3.0-dev

### Platform

GCC 13.3.Rel1 on Windwows

### What happened?

NEMA_GFX does not build since `draw_unit` is undefined.
https://github.com/lvgl/lvgl/blob/5fde4418fb0da9f3c3dab84672eb9f4c3404565f/src/draw/nema_gfx/lv_draw_nema_gfx_label.c#L121C26-L121C35
Am I missing sth?
@liamHowatt, @cristian-stoica

cristian-stoica, 2025-02-03:
Indeed, we're missing a continuous build for Nema that would catch this.

There seems to be some leftovers like draw_unit->target_layer. Please replace them with t->target_layer as in the blame commit. Since you are able to build and test, maybe you can help with a fix? Thanks

tdjastrzebski, 2025-02-03:
Where I work right now I have very poor Internet, loading this page takes several attempts :/

tdjastrzebski, 2025-02-03:
I think it may be a good idea to reverse recent changes until build is fixed.

liamHowatt, 2025-02-04:
I believe it's from #7240. Should be a quick fix.

cc @MarkIoanAmat

tdjastrzebski, 2025-02-05:
I think excluding large parts of code from compilation like `#if LV_USE_NEMA_GFX ...` may not be a good idea, at least not for DEBUG build since it allows for build problems to make it to the master branch undetected while in RELEASE build unused functions are removed anyway.

tdjastrzebski, 2025-02-05:
Is not it too tightly coupled components problem?
I am not saying it is but that is what is sounds like to me.

liamHowatt, 2025-02-12:
Fixed by #7759 or #7718

closed 2025-02-28 by liamHowatt