lvgl/lvgl #7705: button click event fires continuously

tdjastrzebski, 2025-02-04:
### LVGL version

9.3.0.dev c5f88823 2025-02-03 23:59

### Platform

STM32U5G9, GCC 13 on Windows

### What happened?

I am showing that button events LV_EVENT_CLICKED and LV_EVENT_RELEASED fire alternately and continuously when button is kept pressed.
As the result, it is nearly impossible to click button just once.
The issue may be related to #6229 and #7612
Tested with the latest dev code master branch, last commit c5f88823 2025-02-03 23:59

### How to reproduce?

```C
lv_obj_t* btn = lv_button_create(wrapper);
lv_obj_add_event_cb(btn, button_handler, LV_EVENT_CLICKED, (void*)1U);
lv_obj_add_event_cb(btn, button_handler, LV_EVENT_RELEASED, (void*)1U);
lv_obj_t* lbl = lv_label_create(btn);
lv_label_set_text(lbl, "test");

static void button_handler(lv_event_t* e) {
  lv_event_code_t code = lv_event_get_code(e);
  static uint32_t counter = 0;
  printf("button event handler %d %d\n", code, counter);
  counter++;
}
```

AndreCostaaa, 2025-02-04:
Hello,

I can't seem to reproduce the error. 

Here's a screencast of what happens in the simulator:

https://github.com/user-attachments/assets/d9f6bab9-123a-42cf-9939-b2e1e130df45

tdjastrzebski, 2025-02-04:
I think the problem is this: within `indev_proc_press()` `indev->pointer.act_obj` is not being set.
Consequently, on each subsequent `indev_proc_press()` call `indev->prev_state` is set to `LV_INDEV_STATE_RELEASED` and next `indev_proc_release()` is being called where `LV_EVENT_RELEASED` event is sent to the same object which was just clicked.
I may not be the best qualified to fix it.

[image]

[image]

AndreCostaaa, 2025-02-04:
That is interesting, did you try reverting #7612 to see if that commit is what causing it ?

tdjastrzebski, 2025-02-05:
@AndreCostaaa, It turned out to be a well-hidden problem with CTP driver I was provided.
I managed to fix it and now it seems to work like a charm, but it revealed another issue. Now LV_EVENT_CLICKED does not always fire and when both LV_EVENT_RELEASED and LV_EVENT_CLICKED are subscribed to it rarely does. 
I will investigate more and create a new issue if needed. Thanks

closed 2025-02-05 by tdjastrzebski

RogerRU, 2025-02-13:
> [@AndreCostaaa](https://github.com/AndreCostaaa), Оказалось, что это была хорошо скрытая проблема с драйвером CTP, который мне предоставили. Мне удалось ее исправить, и теперь все работает как часы, «визуальные» элементы LVGL работают как положено, но это выявило еще одну проблему. Теперь LV_EVENT_CLICKED не всегда срабатывает, а когда подписаны и LV_EVENT_RELEASED, и LV_EVENT_CLICKED, то срабатывает редко. Я изучу вопрос подробнее и создам новую проблему, если понадобится. Спасибо

Hi!
Could you please tell me more about the problem with the driver and how you solved it.
I have exactly the same problems
ESP32s3 + GT911 (touch driver)