lvgl/lvgl #7706: Glitch in Rotation with LV_DRAW_TRANSFORM_USE_MATRIX Enabled

diylxy, 2025-02-04:
### LVGL version

v9.3.0-dev

### Platform

Visual Studio simulator

### What happened?

Enabling `LV_DRAW_TRANSFORM_USE_MATRIX` and some related settings, then adding rotation to an `lv_obj` causes LVGL to crash with a segmentation fault on Linux. I tried to reproduce this bug on Windows, and the result is shown in the video.
It also affects version 9.2.0.
This rotation transformation works fine without using a matrix.

https://github.com/user-attachments/assets/e8a81e61-704d-474c-8db9-e5f8731b995c

### How to reproduce?

```c
lv_obj_t* lbl;
void anim_cb_rotate(void* var, int val) {
    lv_obj_set_style_transform_rotation((lv_obj_t *)var, val, 0);
    lv_label_set_text_fmt(lbl, "%d", val);
}
void example() {
    lv_obj_t* obj_rotate = lv_obj_create(lv_screen_active());
    lv_obj_set_size(obj_rotate, 300, 300);
    lv_obj_center(obj_rotate);
    // lv_obj_set_style_transform_rotation(obj_rotate, 200, 0);

    lbl = lv_label_create(obj_rotate);
    lv_obj_center(lbl);

    lv_anim_t a;
    lv_anim_init(&a);
    lv_anim_set_var(&a, obj_rotate);
    lv_anim_set_exec_cb(&a, anim_cb_rotate);
    lv_anim_set_time(&a, 20000);
    lv_anim_set_delay(&a, 1000);
    lv_anim_set_repeat_count(&a, LV_ANIM_REPEAT_INFINITE);
    lv_anim_set_values(&a, 0, 3600);
    lv_anim_start(&a);
}
```

liamHowatt, 2025-02-07:
I reproduced it. The problem seems to be with using `transform` style properties when matrix-based transformation is enabled. The `lv_obj_set_transform` is more thoroughly tested for matrix-based object rotation, as in https://docs.lvgl.io/master/examples.html#transform-object-using-a-3x3-matrix .

With layer debug enabled, it looks like the area is being transformed correctly but an extra draw task is being created for the untransformed location and drawing it there, while just drawing a fill in the transformed location.

It only works when the transformed area is clipped by the top of the screen 🤔

cc @FASTSHIFT

https://github.com/user-attachments/assets/712bf82a-ff96-46f7-8896-1882b2c9e60b

FASTSHIFT, 2025-02-08:
`LV_DRAW_TRANSFORM_USE_MATRIX` requires the renderer to support 3x3 matrices (see [here](https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L137)). Currently only the VG-Lite render backend supports this option. If you only have a software renderer, turning this option on will result in rendering errors.

FASTSHIFT, 2025-02-08:
`LV_DRAW_TRANSFORM_USE_MATRIX` works fine on the VG-Lite emulator (text rendering is known to be slightly buggy on the emulator, not on actual hardware)
https://github.com/user-attachments/assets/63cea40d-536e-4d35-b6f4-272b10be1d44

My config file: [lv_conf.h.zip](https://github.com/user-attachments/files/18716800/lv_conf.h.zip)

closed 2025-02-08 by diylxy