lvgl/lvgl #7707: Docs-Build Directory Structure

vwheeler63, 2025-02-04:
### Introduce the problem

Hi, Gábor!  ( @kisvegabor )

I wanted to make this discussion separate rather than mix it in with PR #7597 so as to make our discussion there shorter and easier to read, since this is really a different topic, and thus should be a different task that the one being discussed in the above PR.

I wanted to show you this to see what you think of it.

I am fond of Linux's doc-build directory structure (which I am only distantly familiar with), which uses this structure:
```
./docs/               # documentation source files (a tree structure under it by major doc sections like ours)
./docs/output/        # all generated output goes here
./docs/output/html/   # HTML build output
./docs/output/latex/  # Latex build output
```

And since we still need a temporary directory (due to necessary automation applied via Python), which could be...
```
./docs/tmp/           # temporary directory
```

While `sphinx` does not deal well with having "extra" `.rst` files in the source directory tree, if `sphinx` was told to use `./docs/tmp/` as the source tree, that solves that problem.  Then:

```bash
$ python build.py  clean
```

would first wipe out both `./docs/output/` and `./docs/build_tmp/`.  Or it could be converted to behave like `make clean` and ONLY wipe out those directories, leaving the user to come back later with simply `python build.py`.

Positives:

- keeps "docs" stuff under the "./docs/" directory;
- eliminates the need for additional top-level directories;
- `sphinx` does not mind outputting to a subdirectory of the source directory;
- the Python code that copies source files from `./docs/` to the temp directory can be made to exclude `./docs/output/` and `./docs/tmp/`.

Negatives:

- I do not know what that would break "down the line" that updates the `https://docs.lvgl.io/master/` docs, or whether it would just require editing a `.yaml` file.
- I do not know what else might rely on the `./out_html/` directory.

This is certainly not a "priority", but in my opinion, would be a cleaner setup.

Thoughts?

### Proposal

Included in the above.

kisvegabor, 2025-02-06:
Sounds really good! This way to docs folder would more self-contained and logical.

> I do not know what that would break "down the line" that updates the https://docs.lvgl.io/master/ docs, or whether it would just require editing a .yaml file.

I think we only need to update [this](https://github.com/lvgl/lvgl/blob/master/.github/workflows/compile_docs.yml#L80). 

---

I am ok with the proposed changes Let's go for it!

vwheeler63, 2025-02-06:
> Sounds really good! This way to docs folder would more self-contained and logical.
>
>
> I think we only need to update [...]

Sweet!  Yesterday I did a full scan of all files for that string and that is the only place I found it -- so it would seem to be confirmed!

closed 2025-02-19 by liamHowatt