lvgl/lvgl #7708: Docs-Build Method

vwheeler63, 2025-02-04:
### Introduce the problem

Hi, Gábor!  ( @kisvegabor )

I wanted to make this discussion separate rather than mix it in with PR #7597 and Issue #7707 so as to make our discussion there shorter and easier to read, since this is really a different topic, and thus should be a different task that the one being discussed in those places.

I wanted to show you this to see what you think of it.  This has been in the back of my mind for a long time.

`sphinx` is natively designed to have the user just do:

```bash
$ cd ./docs/
$ make html
$ make latex
$ make clean
etc.
```

and to be honest, the thought of converting the whole docs-generation to use that for all platforms has been in the back of my mind for a long time.  I think most programmers (me included) would appreciate it.  (Since we still need the automation currently being done with Python, the `Makefile`s could simply run Python with the appropriate arguments.)

Thoughts?

### Proposal

Included in the above.

kisvegabor, 2025-02-06:
I think wrapping the python script to `make` just to call `make` instead of python is misleading. If it's a Python script I see no issue with calling it directly as it's clean and transparent.

vwheeler63, 2025-02-06:
> I think wrapping the python script to `make` just to call `make` instead of python is misleading. If it's a Python script I see no issue with calling it directly as it's clean and transparent.

I concur.  After a day of using `python build.py html` and `python build.py clean`, I am convinced that it is super easy and intuitive now.