lvgl/lvgl #7715: Config folder + CI

AndreCostaaa, 2025-02-05:
### Introduce the problem

I don’t know if this has being brought out before but i was thinking that maintaining some configuration files inside a folder like `configs` would be useful for two reasons:

1. Makes it possible for a user to take a config that works and tweak it to suit his needs. Having multiple configs allows the user to have multiple examples to see what is/ isn’t possible
2. It would make it very easy to check for problems like the current broken `nema_gfx` build with CI

It would also be very easy to add new tests to the CI by simply creating a new configuration file

One possible problem is the maintainability issue of keeping multiple configs up to date when a new config arrives but this is mitigated by the CI that will ensure that old configs will still work 

### Proposal

1. Create a `configs` folder and regroup some example configs, the combination of these configs should eventually regroup all of the possible config possibilities inside lvgl

2. Add a ci job that builds lvgl using these configs to make sure they don’t get broken

liamHowatt, 2025-02-12:
I think these configs could be lv_conf.defaults files so they are smaller and clearly show what deviates from lv_conf_template.h

AndreCostaaa, 2025-02-12:
Oh TIL. That looks pretty cool

Do you think it would help having some configs grouped together?

liamHowatt, 2025-03-07:
Yes I think it could be very helpful! There could be some starter configs with good defaults for different types of platforms.

As I think of examples, I think fragments of configs rather than whole configs could also helpful. Not sure.

Some configs / fragments of configs I can think of:
- Bare minimum -- show that there is nothing to fear by leaving everything default -- set `LV_COLOR_DEPTH`
- MPU / libc is available -- set `LV_USE_STDLIB_` to `LV_STDLIB_CLIB`, enable `LV_USE_LOG`, enable `LV_LOG_PRINTF`, set `LV_ASSERT_HANDLER_INCLUDE` to `<assert.h>`, set `LV_ASSERT_HANDLER` to `assert(0);`.
- Multicore -- set `LV_USE_OS` and `LV_DRAW_SW_DRAW_UNIT_CNT`
- Simulator -- Show how to set SDL, X11, Wayland, maybe DRM, maybe FBDEV.
- Filesystem configs for MCU SD card / flash, and for MPU
- Image decoders
- **important** A config that turns on everything needed for VG Lite or ThorVG. Enable matrix, float, vector_graphic, draw_buf_align, draw_buf_stride_align, gpu_init, etc.
- Debug tools - Try one-at-a-time of `LV_USE_REFR_DEBUG`, `LV_USE_LAYER_DEBUG`, and `LV_USE_PARALLEL_DRAW_DEBUG`, and turn on all asserts.
- XML -- Obj name, obj ID, etc.
- Unique RAM arrangement -- `LV_MEM_POOL_EXPAND_SIZE`, `LV_ATTRIBUTE_FAST_MEM`, etc.
- Very low-res I1 display -- `LV_FONT_UNSCII_8`, `LV_COLOR_DEPTH 1`
- Many, many more

liamHowatt, 2025-03-07:
I've been meaning to replace [these lv_conf.h files](https://github.com/lvgl/lv_port_texas_sk-am62p-lp/tree/4efc5718138111de5440f14d47322f27ae8faea4/lvgl_conf_example) with lv_conf.default files. Config folders are useful in port repos like that.

AndreCostaaa, 2025-03-07:
Something that popped into my mind, imagine this workflow:

```bash
cd lv_port_XXX
lv_config minimal_config # renames the old `lv_conf.h` to `lv_conf.h.old` and copies the config called `minimal_config` inside `lvgl/configs` to the current repository
```

`lv_config` could be an app that users could download if they wanted to have this type of workflow

This is usually done with `make` directly and I believe espressif does this with `cmake` but it would be kind of a pain having to add and maintain this in every port, that's why i'm proposing a simple app

Also, I believe having everything on the main repo is easier to maintain, a thing i've noticed is how hard it is to maintain `lv_conf.h` files in port repos, also having this in the main repository would make it easy to add to the lvgl CI directly
Here's an example of a ci that does this inside [so3](https://github.com/smartobjectoriented/so3/blob/1f39bb5b6dc343518a5d5d2dd5e84d253338d325/.github/workflows/build.yml#L11-L29)

closed 2025-04-10 by lvgl-bot