lvgl/lvgl #7728: Key columns not spaced correctly in Keybard numpad

FloHomi, 2025-02-07:
### LVGL version

>=9.2

### Platform

VisualStudio Simulator

### What happened?

when having a keyboard with an linear layout, there is a problem (in maybe the padding?).
[image]
I assume the spacing is a bit larger then you add more buttons.

### How to reproduce?

    lv_obj_t * kb = lv_keyboard_create(lv_scr_act());
    lv_obj_set_style_bg_color(kb, lv_color_hex(0xFF0000), 0);

    lv_keyboard_set_mode(kb, LV_KEYBOARD_MODE_NUMBER);

FloHomi, 2025-02-26:
not stale

FloHomi, 2025-02-26:
not stale

liamHowatt, 2025-03-25:
It's a hard problem because it's technically doing the correct thing. The buttons in the bottom row are all the same width. The logic of the buttonmatrix base widget would need to be changed to align columns. You're welcome to explore the code, try, and open a PR.

To get what you want now, you can create a keypad with a [grid layout](https://docs.lvgl.io/master/details/common-widget-features/layouts/grid.html) and set the `column_span` to `2` for the right-side buttons except for the bottom two.

FloHomi, 2025-03-26:
Thank you Liam for the reply,
I will see what I can do with my limited expertise :-)

Concerning your other suggestion. I assume, you mean to build the keyboard myself, instead of the keyboard widget, right?

liamHowatt, 2025-03-26:
> I assume, you mean to build the keyboard myself, instead of the keyboard widget, right?

Yeah. Apply the grid layout to a normal object and create buttons as children of the object and assign their grid cells. You can see how the columns align in this example: https://docs.lvgl.io/master/examples.html#demonstrate-cell-placement-and-span

FloHomi, 2025-03-26:
Okay Thank you 👍