lvgl/lvgl #7729: indicator Arc is drawn as per not expected.

rahul-navalkar, 2025-02-07:
### Introduce the problem

I am working on one project, where i want to set arc value as per can frame reception. but here arc drawn starting always from 12 o'clock towards both sides. i am attaching video for this. i need arc will drawn from meter value  0 to meter value 8000 as shown in  video. code is :
  lv_obj_t *obj = lv_arc_create(parent_obj);
            objects.archm = obj;
            lv_obj_set_pos(obj, 0, 0);
           lv_obj_set_size(obj, 360, 360);

            lv_arc_set_value(obj, 0);

            lv_arc_set_bg_start_angle(obj, 121);
            lv_arc_set_bg_end_angle(obj, 60);

            lv_obj_set_style_arc_width(obj, 25, LV_PART_MAIN | LV_STATE_DEFAULT);
            lv_obj_set_style_arc_width(obj, 25, LV_PART_INDICATOR | LV_STATE_DEFAULT);
            lv_obj_set_style_arc_color(obj, lv_color_hex(0xffffffff), LV_PART_MAIN | LV_STATE_DEFAULT);
            lv_obj_set_style_arc_color(obj, lv_color_hex(0xffd79513), LV_PART_INDICATOR | LV_STATE_DEFAULT);
            lv_obj_set_style_bg_color(obj, lv_color_hex(0xff63f321), LV_PART_KNOB | LV_STATE_DEFAULT);
            lv_obj_align(obj, LV_ALIGN_CENTER, 0, 0); // rahul
            lv_arc_set_rotation(obj, 0); 

### Proposal

https://github.com/user-attachments/assets/989877de-c24e-48f7-8717-a9bf7e8b8547

rahul-navalkar, 2025-02-07:
@kisvegabor  sir help me out.

lizhaoming634, 2025-02-07:
https://github.com/user-attachments/assets/9c1e4e8f-b52c-461f-8f90-ec3a93bfdb59
static void set_angle(void * obj, int32_t v)
{
    lv_arc_set_value(obj, v);
}

void lv_example_bar_1(void)
{
    lv_obj_t *obj = lv_arc_create(lv_scr_act());

    lv_obj_set_pos(obj, 0, 0);
    lv_obj_set_size(obj, 360, 360);

    lv_arc_set_value(obj, 0);

    lv_arc_set_bg_start_angle(obj, 121);
    lv_arc_set_bg_end_angle(obj, 60);

    lv_obj_set_style_arc_width(obj, 25, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_arc_width(obj, 25, LV_PART_INDICATOR | LV_STATE_DEFAULT);
    lv_obj_set_style_arc_color(obj, lv_color_hex(0xffffffff), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_arc_color(obj, lv_color_hex(0xffd79513), LV_PART_INDICATOR | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_color(obj, lv_color_hex(0xff63f321), LV_PART_KNOB | LV_STATE_DEFAULT);
    lv_obj_align(obj, LV_ALIGN_CENTER, 0, 0); // rahul
    lv_arc_set_rotation(obj, 0);

    lv_anim_t a;
    lv_anim_init(&a);
    lv_anim_set_var(&a, obj);
    lv_anim_set_exec_cb(&a, set_angle);
    lv_anim_set_time(&a, 1000);
    lv_anim_set_repeat_count(&a, LV_ANIM_REPEAT_INFINITE);    /*Just for the demo*/
    lv_anim_set_repeat_delay(&a, 500);
    lv_anim_set_values(&a, 0, 100);
    lv_anim_start(&a);

}

rahul-navalkar, 2025-02-26:
i am not able to resolve this issue, if anybody solution, please provide.

kisvegabor, 2025-03-03:
Hi,

Sorry for the long delay. It seems it happens because your display is very slow. You can just call delay 3 secs after startup and call `lv_obj_clean(lv_screen_active())`. I believe you will see that the screen is screen fro top to bottom very slowly.

rahul-navalkar, 2025-03-05:
Thanks sir for your reply. I will try this.

rahul-navalkar, 2025-03-10:
@kisvegabor  i tried your solution, still arc is not filled as per expectation.  Please correct me if I am wrong. Here is code sample:

[image]

rahul-navalkar, 2025-03-12:
this is reference.
https://github.com/user-attachments/assets/3de537a8-2d42-4a46-882b-219b51fdf29a

rahul-navalkar, 2025-03-15:
ok I will try to increase SPI speed.

closed 2025-04-10 by lvgl-bot