lvgl/lvgl #7738: Alternate arc_width settings cause memory leak

joeltamkin, 2025-02-08:
### LVGL version

9.2.2

### Platform

ESP-IDF 5.3.2 / lvgl_lvgl 9.2.2 as component from ESP respository. Running on ESP32-S3 hardware

### What happened?

Setting arc width to something other than default (in this case '20') causes lvgl memory use to climb with each subsequent screen redraw.  This happens regardless of whether the arc indicator value or other properties are changed.  If it's being redrawn in any way, an additional 380-400 bytes of memory are consumed (as reported by lv_mem_monitor()) with each screen update.

### How to reproduce?

Code snippets - part of a much larger project, but isolated to the style adjustments shown:
```c
        lv_obj_t * LVObj_positionIndicator;
	lv_style_t LVStyle_positionIndicator;
        lv_obj_t * LVObj_boostGauge;
        lv_style_t LVStyle_boostGauge;
        lv_style_t LVStyle_boostGaugeIndicator;

        static lv_mem_monitor_t mon;
```
```c
        LVObj_positionIndicator = lv_label_create(scr);
	lv_style_init(&LVStyle_positionIndicator);
	lv_style_set_text_font(&LVStyle_positionIndicator, &lv_font_dsdigital_100);
	lv_style_set_text_color(&LVStyle_positionIndicator, lv_color_white());
	lv_obj_align(LVObj_positionIndicator, LV_ALIGN_CENTER,30,0);
	lv_style_set_text_align(&LVStyle_positionIndicator, LV_TEXT_ALIGN_CENTER);
	lv_label_set_text_static(LVObj_positionIndicator, racePositionText);
	lv_obj_add_style(LVObj_positionIndicator,&LVStyle_positionIndicator, LV_PART_MAIN);
        ...
        LVObj_boostGauge = lv_arc_create(scr);
	lv_style_init(&LVStyle_boostGauge);
	lv_style_init(&LVStyle_boostGaugeIndicator);
	lv_obj_add_style( LVObj_boostGauge, &LVStyle_boostGauge, LV_PART_MAIN);
	lv_obj_add_style( LVObj_boostGauge, &LVStyle_boostGaugeIndicator, LV_PART_INDICATOR);
	lv_obj_set_size(LVObj_boostGauge, 230, 230);
	lv_obj_center(LVObj_boostGauge);
	
	lv_arc_set_rotation(LVObj_boostGauge, 150);
	lv_arc_set_bg_angles(LVObj_boostGauge, 0, 240);
	
	lv_style_set_arc_color(&LVStyle_boostGauge, lv_color_hex(UNINITIALIZEDWIDGETCOLOR));
	lv_style_set_arc_color(&LVStyle_boostGaugeIndicator, lv_color_hex(0xFF0000));
	
	//lv_obj_set_style_arc_width(LVObj_boostGauge, 20, LV_PART_MAIN);
	//lv_obj_set_style_arc_width(LVObj_boostGauge, 20, LV_PART_INDICATOR);
	// (EITHER OF THESE FORMS, ABOVE OR BELOW, EXPOSE THE REPORTED BUG)
	lv_style_set_arc_width(&LVStyle_boostGauge, 20);
	lv_style_set_arc_width(&LVStyle_boostGaugeIndicator, 20); 	 	
		
	lv_arc_set_range(LVObj_boostGauge,0,100);
	lv_arc_set_value(LVObj_boostGauge,50);
	lv_obj_remove_style(LVObj_boostGauge, NULL, LV_PART_KNOB);
```
Implementing updates to the label text will cause memory usage to increase with each redraw:
```c 
        lv_mem_monitor(&mon);
	printf("Pre-set_text used: %6d (%3d %%), frag: %3d %%, biggest free: %6d\n", (int)mon.total_size - mon.free_size, mon.used_pct, mon.frag_pct, (int)mon.free_biggest_size);
        if (lvgl_port_lock(0)) {
             lv_label_set_text_static(LVObj_positionIndicator, "1");
        }
        lvgl_port_unlock();
        vTaskDelay(500) // some wait necessary for mem_monitor to show result of redraw...
        lv_mem_monitor(&mon);
	printf("POST-set_text used: %6d (%3d %%), frag: %3d %%, biggest free: %6d\n", (int)mon.total_size - mon.free_size, mon.used_pct, mon.frag_pct, (int)mon.free_biggest_size);
```
Console output snippet below:
```
Pre-set_text used:  27328 ( 45 %), frag:   5 %, biggest free:  32784
POST-set_text used:  27716 ( 45 %), frag:   4 %, biggest free:  32784
Pre-set_text used:  27716 ( 45 %), frag:   4 %, biggest free:  32784
POST-set_text used:  28112 ( 46 %), frag:   6 %, biggest free:  31620
Pre-set_text used:  28112 ( 46 %), frag:   6 %, biggest free:  31620
POST-set_text used:  28508 ( 47 %), frag:   7 %, biggest free:  30988
Pre-set_text used:  28508 ( 47 %), frag:   7 %, biggest free:  30988
POST-set_text used:  28912 ( 47 %), frag:   6 %, biggest free:  30988
```
This increase is not seen when arc width properties for the separate boost gauge object are not set.  Memory use will grow until lvgl memory allocation is reached.

joeltamkin, 2025-02-08:
Also verified that setting the arc width to 10 (default value) causes the issue as well.

lizhaoming634, 2025-02-09:
[image]
Alternatively, you can use lv_obj_set_style_arc_width

joeltamkin, 2025-02-09:
I don't want to reset the style. I want the width I specified when creating the arc. This is not about changing or resetting the style of the arc. As you can see from the code snippets, I tried both methods (style and local object) of setting arc width and had the same behavior.