lvgl/lvgl #7739: I am confused about how to connect LVGL9 to EPDIY (e-link)?

lanistor, 2025-02-09:
### LVGL version

v9.2

### Platform

ESP32S3

### What happened?

I am using LVGL9 to connect to the ink screen through EPDIY. I used LVGL8 to connect successfully before. After upgrading to LVGL9, I used LV_COLOR_FORMAT_RGB888 mode to connect successfully, but each pixel will take up 3 bytes, so I tried to optimize.

Tip: I have set COLOR_DEPTH=1

I tried to set LV_COLOR_FORMAT_I8, LV_COLOR_FORMAT_L8, LV_COLOR_FORMAT_I4 mode, but they all failed to render successfully. The entire screen is almost completely black with a little bit of white.
I am confused and have not been able to succeed. 

I also tried to render using LV_COLOR_FORMAT_I1, and it can be rendered at 1/8 size, which is in line with expectations.

My questions are:
1. How should COLOR_DEPTH and color_format be set, and then how to convert colors to epdiy buffer?
2. I set COLOR_DEPTH=1, why can I still read non-0x00, 0xff colors in the lvgl buffer? For example, 0x14, 0xa3, etc. Doesn't COLOR_DEPTH=1 limit it to only 0x00 or 0xff?

### How to reproduce?

This is my code for converting colors to epdiy (For LV_COLOR_FORMAT_I8 or LV_COLOR_FORMAT_L8), this will run on `flush_cb`:
```cpp
void buf_copy_to_framebuffer(EpdRect image_area, uint8_t* image_data) {
  assert(framebuffer != NULL);

  auto display_width  = epd_rotated_display_width();
  auto display_height = epd_rotated_display_height();
  auto count = 0;
  for (uint32_t i = 0; i < image_area.width * image_area.height; i++) {
    uint8_t val = image_data[i]; //  > 0x99 ? 0xff : 0x00;
    // ets_printf(" 0x%2x ", image_data[i]);

    int xx = image_area.x + i % image_area.width;
    if (xx < 0 || xx >= display_width) {
      continue;
    }
    int yy = image_area.y + i / image_area.width;
    if (yy < 0 || yy >= display_height) {
      continue;
    }
    uint8_t* buf_ptr = &framebuffer[yy * display_width / 2 + xx / 2];
    if (xx % 2) {
      *buf_ptr = (*buf_ptr & 0x0F) | (val << 4);
    } else {
      *buf_ptr = (*buf_ptr & 0xF0) | (val >> 4);
    }
  }
}
```

And here is my init code: 
```cpp
// 应用入口
void app_main() {
  hal_init();

  int stack_size = 1024 * 28;
  xTaskCreatePinnedToCore(gui_task, "gui_task", stack_size, NULL, 5,
                          get_gui_task_handler(), 0);
}

void hal_init(void) {
  ESP_LOGI(TAG, "gui_task hal_init started");

  lv_init();

  lvgl_driver_init();

  lv_tick_set_cb(sys_tick);

  int display_width  = epd_rotated_display_width();
  int display_height = epd_rotated_display_height();

  disp = lv_display_create(display_width, display_height);
  lv_display_set_flush_cb(disp, disp_driver_flush);
  lv_display_set_color_format(disp, LV_COLOR_FORMAT_I8);

  lv_color_format_t cf = lv_display_get_color_format(disp);
  uint32_t px_size = lv_color_format_get_size(cf);

  ESP_LOGI(TAG, "epd_display_width: %d; epd_display_height: %d; "
      "sizeof(lv_color_t): %d, LV_COLOR_DEPTH: %d, px_size: %d",
           display_width, display_height, sizeof(lv_color_t), LV_COLOR_DEPTH, px_size);

  int display_pixel_size = display_width * display_height / 2;
  static lv_color_t* buf1 = (lv_color_t*)heap_caps_malloc(
    display_pixel_size * px_size, MALLOC_CAP_SPIRAM);
  static lv_color_t* buf2 = (lv_color_t*)heap_caps_malloc(
    display_pixel_size * px_size, MALLOC_CAP_SPIRAM);
  assert(buf1 != NULL);
  assert(buf2 != NULL);

  lv_display_set_buffers(disp, buf1, buf2, display_pixel_size * px_size, LV_DISPLAY_RENDER_MODE_PARTIAL);
  
  lv_group_set_default(lv_group_create());

  lv_indev_t* indev = lv_indev_create();
  lv_indev_set_type(indev, LV_INDEV_TYPE_POINTER);
  lv_indev_set_read_cb(indev, touch_driver_read);
  lv_indev_set_long_press_time(indev, 600);
}

static void gui_task(void* arg) {
  fl_before_start();

  start_application();

  fl_after_start();

  /* Force screen refresh */
  lv_refr_now(NULL);

  while (1) {

     uint32_t time_till_next = 10;
    if (pdTRUE == xSemaphoreTake(gui_semaphore, portMAX_DELAY)) {
     uint32_t _next_time = lv_timer_handler();
     if (_next_time < time_till_next) {
          time_till_next = _next_time;
      }
      xSemaphoreGive(gui_semaphore);
    } else {
      // ets_printf(TAG " lv_task_handler error \n");
    }

    if(gui_task_queue_check()) {
      time_till_next = 0;
    }

    if (time_till_next > 0) {
      vTaskDelay(time_till_next / portTICK_PERIOD_MS);
    }
  }

  vTaskDelete(NULL);
}
static uint32_t sys_tick(void) {
  return (uint32_t) (esp_timer_get_time() / 1000ULL);
}

```

lanistor, 2025-02-25:
May i get some suggestion?

tsl0922, 2025-03-21:
> I set COLOR_DEPTH=1, why can I still read non-0x00, 0xff colors in the lvgl buffer? For example, 0x14, 0xa3, etc. Doesn't COLOR_DEPTH=1 limit the color value only in [0x00, 0xff]?

No. 1bit color means a pixel uses 1bit in lvgl buffer.

Here is the working code that I used to draw lvgl buffer to e-paper (with `LV_COLOR_DEPTH=1`):

```c
void drawBitmap(const uint8_t bitmap[], int16_t x, int16_t y, int16_t w, int16_t h, uint16_t color)
{
  int16_t byteWidth = (w + 7) / 8;
  uint8_t byte = 0;
  for (int16_t j = 0; j < h; j++) {
    for (int16_t i = 0; i < w; i++ ) {
      if (i & 7) byte <<= 1;
      else {
        byte = bitmap[j * byteWidth + i / 8];
      }
      if (!(byte & 0x80)) {
        drawPixel(x + i, y + j, color);
      }
    }
  }
}
```

closed 2025-04-24 by lvgl-bot