lvgl/lvgl #7746: LVGLImage.py only supports png images?

Dozingfiretruck, 2025-02-10:
### Introduce the problem

LVGLImage.py only supports png images?
Where can I see the source code conversion that https://lvgl.io/tools/imageconverter v9 ?
thanks

Berte-Yu, 2025-02-14:
[LVGLImage.py](https://github.com/lvgl/lvgl/blob/master/scripts/LVGLImage.py)

Dozingfiretruck, 2025-02-14:
> [LVGLImage.py](https://github.com/lvgl/lvgl/blob/master/scripts/LVGLImage.py)

Using LVGLImage.py to process image files other than png will prompt FormatError: PNG file has invalid signature.

Dozingfiretruck, 2025-03-06:
> We need some feedback on this issue.
> 
> Now we mark this as "Abandoned" because there was no activity here for 14 days.
> 
> Remove the "Stale" label or comment else this will be closed in 7 days.

do not close

liamHowatt, 2025-03-07:
Yes, PNG is the only standard image format that it supports for now. What type of image are you trying to use?

Dozingfiretruck, 2025-03-08:
> Yes, PNG is the only standard image format that it supports for now. What type of image are you trying to use?

I make a gui toot：
https://github.com/Dozingfiretruck/lvgl_conv_tool
so ，I want to it support all types，like web

liamHowatt, 2025-03-10:
@XuNeo could it use PIL/Pillow to load the images?

XuNeo, 2025-03-11:
Hi Liam,

Yes, PIL/Pillow is able to load JPG. What I'm not certain is how to generate indexed image, since currently `pngquant` is used to generate the palette for best quality.

The current `LVGLImage.py` is not good enough to extend for more image format. @W-Mai has written a rust image converter tool, which is a good alternative. See https://github.com/W-Mai/icu

liamHowatt, 2025-03-11:
Thank you XuNeo!

Dozingfiretruck, 2025-03-13:
Thank you，so Is there a plan to support all formats in LVGLImage.py later on?

closed 2025-04-10 by lvgl-bot