lvgl/lvgl #7750: How to make the LVGL simulator to display a round screen?

Pzhaowei, 2025-02-11:
### Introduce the problem

How to make the LVGL simulator to display a round screen?

### Proposal

[image]

rahul-navalkar, 2025-02-11:
can you provide detailes like which display controller and size of display?

Pzhaowei, 2025-02-11:
> can you provide detailes like which display controller and size of display?

windows LVGL simulator and arbitrary resolution(such as 320x320). Now only a square screen like below, but I want to a round screen.

[image]

lizhaoming634, 2025-02-11:
can you try lv_obj_set_style_radius(lv_screen_active(), LV_RADIUS_CIRCLE, LV_PART_MAIN);?

rahul-navalkar, 2025-02-11:
basically I used squareline studio and eez studio, they provide ui for screen design in which we can enable circular display.

fbiego, 2025-03-07:
You can make your screen object rounded using the below;

```
lv_obj_set_style_radius(lv_screen_active(), LV_RADIUS_CIRCLE, 0);
lv_obj_set_style_clip_corner(lv_screen_active(), true, 0);
```

liamHowatt, 2025-03-07:
This is interesting because clipping the corners in the simulator is good for development, but you should remember to disable it on-device because it adds to the rendering time unnecessarily, when the display is already physically clipped 😉

closed 2025-04-10 by lvgl-bot