lvgl/lvgl #7752: LVGL does not render animated SVGs

pragulka, 2025-02-11:
### LVGL version

v9.3.0-dev

### Platform

SDL + Win11, ESP32S3

### What happened?

According to [doc](https://docs.lvgl.io/master/details/libs/svg.html#usage) SVG parser handles animation attributes, but LVGL renders only a static image.

[image]

### How to reproduce?

```
static void event_cb(lv_event_t * e)
{
    static char svg_data[] = R"(<svg width="200" height="180" xmlns="http://www.w3.org/2000/svg">
            <rect x="30" y="30" height="110" width="110" fill="red" stroke="blue" stroke-width="10">
              <animateTransform
                attributeName="transform"
                begin="0s"
                dur="10s"
                type="rotate"
                from="0 85 85"
                to="360 85 85"
                repeatCount="indefinite" />
            </rect>
          </svg> )";
    lv_layer_t * layer = lv_event_get_layer(e);
    lv_svg_node_t * svg = lv_svg_load_data(svg_data, sizeof(svg_data) / sizeof(char));
    lv_draw_svg(layer, svg);
    lv_svg_node_delete(svg);
}

void lv_example_svg_1(void)
{
    lv_obj_add_event_cb(lv_screen_active(), event_cb, LV_EVENT_DRAW_MAIN, NULL);
}
```

onecoolx, 2025-02-18:
Well, parsing of SVG animation attributes is currently supported. But if we need to play SVG animations, we need to develop an SVG animation player. Ideally, it should holded the SVG dom tree, read the animation attribute information, and use a timer to drive the animation playback, just like the gif widget. No such component has been developed yet.

J-u-n-o, 2025-02-19:
Note1:
Maybe add this to the lv_conf_template.h?
`// Enable does not animate SVG: this is not (yet) implemented`
`#define LV_USE_SVG_ANIMATION 0`

Note2:
LVGL uses ThorVG is used to play Lottie animations which functions.
ThorVG can also parse and show SVG; one needs to addapt lv_lottie to support mimetype "svg". But ThorVG only does static SVG.
See:
https://github.com/thorvg/thorvg/issues/3187

Trying to create a lv_svg based on lv_lottie, but it crashes (realloc failed) on rleRender _horizLine

closed 2025-03-28 by lvgl-bot