lvgl/lvgl #7754: LVGL X11 deletion missing termination of the X11 connection.ll

phelter, 2025-02-11:
### LVGL version

v9.2.0

### Platform

VSCode on both a Ubuntu 24.04 machine and Windows machine using WSL with a docker image of Ubuntu 24.04.
I have a unit test suite using GoogleTest to run and execute the LVGL code using X11 driver to view the contents of various locally generated widgets.

### What happened?

When performing unit tests with lvgl code and constantly starting-up and shutting down an X11 display many times - eg:

`testApp --gtest_repeat=1000 --gtest_throw_on_failure --gtest_shuffle` when the testApp starts and stops an X11 display, I encounter it terminating incorrectly every so often.

Once it fails once, the application will not re-open an X11 connection or window again until the system is reboot (or in the case of the docker image it is shut down and start up again).

### How to reproduce?

I've tracked it down to the following code: https://github.com/lvgl/lvgl/blob/43b32341b09724816c73deb29a6bc66b328b8f2b/src/drivers/x11/lv_x11_display.c#L211

The window is correctly destroyed and flushed, however the Display connection is NOT terminated correctly.

The above code needs to add: [XCloseDisplay()](https://www.x.org/releases/current/doc/libX11/libX11/libX11.html#Closing_the_Display) as per the Xlib documentation to properly close the connection to X11 before terminating.

```c
    XFlush(xd->hdr.display);
    XCloseDisplay(xd->hdr.display);
    
    lv_free(xd);
```

To reproduce create a simple application using google_test with X11 as the display type and have it create a basic black screen and then startup and then shut down.  I was able to produce it with this + a simple label and changing the label text every so often with the testing.  You MUST run many many times as mentioned above using `testApp --gtest_repeat=1000 --gtest_throw_on_failure --gtest_shuffle` - the gtest_shuffle is not as important but eventually the app will stop running before the repeat is complete and the resulting X11 will not be usable, despite a Ctrl-C exit of the application.

phelter, 2025-02-11:
Here is a git patch file to fix the issue.

I've run it several times with over 10000 runs of various tests and it does not fail now.

[x11.patch](https://github.com/user-attachments/files/18759653/x11.patch)

liamHowatt, 2025-03-07:
I opened a PR: #7904. Next time feel free to open one yourself. It will be addressed quicker, although we have a goal of addressing issues within 1 week which we are working towards achieving consistently.

I tried with `while :; do timeout -s SIGINT 0.1 ./lvgl_workspace ; done` to rapidly launch and ctrl-c the process but I don't see a difference with or without your patch, and no sign of the critical crash. I think it's fine to merge your patch anyways as it fixes yours, and has no ill effects otherwise 🙂

Both with and without the patch I see
```
[Warn]  (0.025, +25)     x11_event_handler: unhandled x11 event: 18 lv_x11_display.c:285
```

phelter, 2025-03-07:
Thanks @liamHowatt - I have a test case that fails without it and passes with it after 100->1000 runs with the exact change you made.  Much appreciated.

PR #7904 (2025-03-17): fix(X11): call XCloseDisplay in delete
Fixes #7754

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/code-format.py` (`astyle v3.4.12` needs t [... cut]
closed 2025-03-17 by kisvegabor