lvgl/lvgl #7760: Drawing labels / letters asynchronous

cosmindanielradu19, 2025-02-12:
### Introduce the problem

During the changes made for NXP's VGLite implementation, there was added an asynchronous task dispatching mechanism for vglite render thread.

On RT1170, where the CPU is faster than the GPU, we have observed that the letters that are not being drawn correctly.

[Here](https://github.com/lvgl/lvgl/blob/master/src/draw/lv_draw_label.c#L582) for each letter, the same buffer is used, unless the buffer is too small, when it's reshaped. The issue that we have is that on our side, there are multiple letters prepared and submitted to the GPU. For example, if the GPU is busy with a task, multiple letters are prepared in the command buffer, and when the GPU will be on idle state, it will start processing a batch of multiple letters. 

The problem is that there doesn't exist a buffer for each letter that should be drawn, so the address that is sent to the GPU might be the same for multiple letters, and that memory area might be written by the CPU when it's preparing another letter while the GPU is reading from that address.

We've thought to multiple solutions for this:
1. A dynamic buffer for each letter
- this solution fixes the issue presented above
- allocating and deallocating the buffers take time and the performance is lower
2. A pool of buffers for letters that are being reused
- not implemented / tested
3. Initializing the fonts as static buffers when LVGL is being initialized and just send the address of each letter to the GPU
- not implemented / tested

What do you think about those solutions? Do you have any alternative to this?

kisvegabor, 2025-02-14:
In the other (generic) VG-Lite driver @FASTSHIFT uses list to copy and save all the draw_bufs, and free them when the command buffer is ready. 

However before going too far we should take 2 things into account:
1. @uLipe is working on unifying the 2 VG-Lite driver. It's not easy as it seems there are differences in the NXP and the  official VG-Lite kernel drivers, but @uLipe can tell more about the details.  Once he has a solid ground and both VG-Lite drivers are working e.g. on NXP i.MX RT 1170 we start the actual unification can start. For this we can pick the best parts of both drivers. What I know is the generic driver support
  - LVGL vector API
  - handles the letter buffering (the problem described in this issue)
But we will need a joint effort to understand how the get the best parts of each. 
2. @FASTSHIFT has an almost ready PR to make font caching simpler. See #7407

nicusorcitu, 2025-02-14:
We should unify our effort into a single VGLite implementation. It does not matter who is the originator as long as the performance is better.

cosmindanielradu19, 2025-03-05:
Not stale

kisvegabor, 2025-03-06:
My solution here would fix it too: https://github.com/lvgl/lvgl/pull/7407#issuecomment-2703262540

closed 2025-04-09 by lvgl-bot