lvgl/lvgl #7766: Buffer Allocation Fail

NehalAR, 2025-02-13:
### LVGL version

v9.2.2

### Platform

Arduino IDE, esp32-wroom32e N8R8, ST7789, using lv_conf.h

### What happened?

Buffer allocation is failing.

Here are the logs: [ESP32-WROOM32E with PSRAM 8MB]

> **Free heap: 2230812**
**Largest block: 2031604**
**MALLOC_CAP_8BIT      2230812 bytes**
**MALLOC_CAP_DMA        176684 bytes**
**MALLOC_CAP_SPIRAM    2054128 bytes**
**MALLOC_CAP_INTERNAL   244188 bytes**
**MALLOC_CAP_DEFAULT   2230812 bytes**
**MALLOC_CAP_IRAM_8BIT       0 bytes**
**MALLOC_CAP_RETENTION       0 bytes**
[Warn]	(0.940, +940)	 lv_draw_buf_create_ex: No memory: 236x100, cf: 16, stride: 944, 94400Byte,  lv_draw_buf.c:296
[Warn]	(0.950, +10)	 lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:442
[Warn]	(0.959, +9)	 lv_draw_buf_create_ex: No memory: 236x100, cf: 16, stride: 944, 94400Byte,  lv_draw_buf.c:296
[Warn]	(0.969, +10)	 lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:442
[Warn]	(0.979, +10)	 lv_draw_buf_create_ex: No memory: 236x100, cf: 16, stride: 944, 94400Byte,  lv_draw_buf.c:296

This goes on and on the screen looks like this:

[image]

Interesting behaviors:

* `ui_EndGong1_screen_init();` works only when `DRAW_BUF_SIZE is (TFT_HOR_RES * 40)` not more or less.

* `ui_MeditationScreen_screen_init();` doesn't work on any buffer size.

*  `ui_BluetoothScreen_screen_init();` works on both when buffer size is `DRAW_BUF_SIZE (TFT_HOR_RES * 40) or DRAW_BUF_SIZE is (1024 * 40)`

### How to reproduce?

It's a rough sketch to check how individual screens look like:
[LVGL_Arduino.zip](https://github.com/user-attachments/files/18781261/LVGL_Arduino.zip)

NehalAR, 2025-02-13:
@kisvegabor  Also tested dual buffers but that doesn't effect anything.

closed 2025-02-18 by NehalAR

kisvegabor, 2025-02-25:
Hi @NehalAR,

If you scale only images, using `lv_image_set_scale` would be better as it doesn't allocate a layer.