lvgl/lvgl #7787: lv_bin_decoder produces build warnings when decompressing LZ4 compressed images

chrismapp, 2025-02-17:
### LVGL version

v9.2.2

### Platform

- ESP32-S3 using ESP-IDF plugging within Visual Studio Code.
- LV_USE_LZ4 is enabled
- Using images compressed using LZ4

### What happened?

In `lv_bin_decoder.c` within the `decompress_image()' function, one of the print format specifiers is incorrect. `len`, which is an `int` uses the `LV_PRId32' format specifier:

https://github.com/lvgl/lvgl/blob/589e8ac994dbdbe1729fce0a865106e94ee5b8c3/src/libs/bin_decoder/lv_bin_decoder.c#L1155

As a result the following warning (treated as an error) is produced:

```
managed_components/lvgl__lvgl/src/libs/bin_decoder/lv_bin_decoder.c:1138:25: error: format '%ld' expects argument of type 'long int', but argument 7 has type 'int' [-Werror=format=]
 1138 |             LV_LOG_WARN("Decompress failed: %" LV_PRId32 ", got: %" LV_PRId32, out_len, len);
      |                         ^~~~~~~~~~~~~~~~~~~~~~                                          ~~~
      |                                                                                         |
      |                                                                                         int
```

A simple fix would be to change the line to:

```
 LV_LOG_WARN("Decompress failed: %" LV_PRId32 ", got: %d", out_len, len);
```

However, I believe that there are other issues in the `decompress_image()` function that it might be good to address at the same time. I believe for example that `out_len` is also using the wrong format specifier, but this is not flagged because the variable is cast to void with the `LV_UNUSED()` macro. In addition to this, there is the opportunity to reduce some repeated code.

The following relatively small change could we applied as an alternative to the above:
https://github.com/lvgl/lvgl/commit/7fd1351e40fe2beee6332b8ece3b5937092aeed4

Or a more substantial change, with additional refactoring could be applied. I have prepared a branch, which can be found here:
https://github.com/chrismapp/lvgl/tree/bugfix-lv-bin-decoder-incorrect-print-format-specifier

I'd be happy to raise a pull request, but wanted to get some feedback on what would be the most appropriate way to fix the issue before I do.

ev0rtex, 2025-02-20:
Just came here to report the same issue... I'm glad I'm not the only one seeing this issue. I apparently had this issue even with 9.2.0 and forgot that I had added a manual typecast to my code since I just had to get it to build at the time but then it resurfaced after I updated to 9.2.2 and I had forgotten about that fact.

chrismapp, 2025-02-20:
I can put out a pull request (although I need to check if there are pre-requisites to doing this, as I have not previously contributed). The three options I proposed were:

1. Do minimum. Just change the print format specifier to %d
2. Use a intermediate variable for the return value of LZ4_decompress_safe() but don't refactor further (see https://github.com/lvgl/lvgl/commit/7fd1351e40fe2beee6332b8ece3b5937092aeed4)
3. A bit more of a refactor (see the prepared change on this branch: https://github.com/chrismapp/lvgl/tree/bugfix-lv-bin-decoder-incorrect-print-format-specifier)

I am not familiar with what testing is in place, so a little unsure how safe it is to refactor.

Can anyone advise on the best approach? Maybe just put a pull request out for 3) and see what comments I receive?

liamHowatt, 2025-03-14:
Brilliant! Yes, the bad `LV_UNUSED`s should be removed. I couldn't spot bad format specifiers for `input_len` or `output_len`, though.

(1.) for sure. (3.) really is a good refactor so I think it can be the PR if you can! Thanks!

I will do a proper review in the PR but I see that `method` in `LV_LOG_WARN("Unknown compression method: %d", compressed->method);` is `uint32_t` so the format specifier should be changed.

chrismapp, 2025-04-13:
Adding a comment to keep open. I will look to raise a pull request from the linked branch above, or happy for someone else to do this. At the moment I can't work on open source work until my current employment ends in early May.

ev0rtex, 2025-05-03:
Also commenting to keep this open. I haven't been working on any of my LVGL projects for a bit but I'm about to start on a new one where I'm certain I'll run into this again. If nobody else gets to it before then I'll see if I can get a PR put together with a proper fix based on the previous comments here.

PR #8234 (2025-05-13): fix(bin_decoder): fix build warnings when decompressing LZ4 compressed images
Fixes #7787 

- Change print format specifiers and use intermediate variable to set decompression length
- Refactor to remove repeated code and improve readability
closed 2025-05-13 by uLipe