lvgl/lvgl #7801: VGLite client API does not have the same behavior when running on a particular plataform.

uLipe, 2025-02-19:
### Introduce the problem

As we already know the LVGL has two implementations of the VGLite API specification, although the implementations can be different (this is another problem when we move towards unification), the results of execution given a particular GPU supported  like gc355 or gc255 should be the same for example running the LVGL Benchmark demo, but considering the following test scenario:

* Board: IMX-RT1170-EVKB + IHM add on board;
* LVGL: master branch;
* Verisilicon latest VGLite kernel driver;
* This project branch: https://github.com/lvgl/lv_port_nxp_imxrt1170-evkb/tree/feature/vglite
* MCU Expresso IDE: Latest version

I was able to swap the VGLite implementations by simply changing the lv_conf.h, but the results observed were quite different, using the specific NXP VGLite client API the Benchmark Demo runs as expected, but when changing from it to the non-NXP VGLite implementation, only the containers and the first scene gets rendered correctly, the text and image scenes gets corrupted until the arc scene that falls into an assertion of invalid argument.

During the comparison between the two implementations, I started to wondering what is missing on the non-NXP implementation to get it is running correctly (consider a GC355 present on IMX), the only file that I still have some guess is the vg_lite_options.h on the VGLite kernel driver which points what is present and what is not present given a particular platform. Would the VGLite implementations consume this file in a different ways?

I'd like to see your visions regarding that.

cc: @FASTSHIFT @XuNeo @nicusorcitu , @cosmindanielradu19 , @junminacts , @kisvegabor

kisvegabor, 2025-02-20:
The root of the differences can be if @FASTSHIFT doesn't use NXP chips. In this case there might be differences in the used GPU (GCxxx) or there can be other vendor specific issues. 

I tried to phrase the above to not disclose any info about the chips used at Xiaomi, however if you can name the chip vendor it'd simplify the discussion. 🙂

FASTSHIFT, 2025-02-21:
This is the configuration of the GPU simulator environment that I usually debug. It is closer to our hardware environment. You can compare this configuration.
[lv_conf.h.zip](https://github.com/user-attachments/files/18900355/lv_conf.h.zip)

uLipe, 2025-02-21:
Thank you @FASTSHIFT  I will try your config environment on the current hardware I have.

nicusorcitu, 2025-02-24:
We need to align on using the same VgLite driver version. NXP uses 4.0.90 currently.

NXP implementation on LVGL 9 works with VgLite 4.x driver. I know there have been some API breakage since VGLite 3.x to 4.x so I hope this is not the case you are running into. AFAIK Xiaomi also uses 4.x (not sure on exact version).

PR #8014 (2025-07-15): feature(draw): vglite: initial refactor to unify the vglite API
Fixes #7801 #7348

This is the initial effort to unify the VGLite client API present on LVGL.

For selecting the direction I made the LVGL benchmark to run into simulator using ThorVG and on the hardware using IMXRT1176 - EVKB and the results are:

- The `draw/nxp/vgilte` implementation works as intended on both cases;
- The  `draw/vg_lite` implementation works only partially on the hardware, the text rendering raises VG_LITE ASSERT, and the following scenes after this does not render correctly.
- The draw dispatcher from `draw/nx [... cut]