lvgl/lvgl #7803: Location of Documentation Images

vwheeler63, 2025-02-19:
### Introduce the problem

Is there any reason that most of the documentation images are in `./docs/misc/...` ?

Sphinx documentation (so we assume this is the designer's intention) puts images used in documentation in the `./doc/_static/` with sub-directories where images are grouped by theme.  But the non-theme images are directly in `./doc/_static/`.  

Does anyone have any objections to the images being moved to our `./docs/_static/` or `./docs/_static/images/` so they are all in one place?

Kind regards,
Vic

### Proposal

My vote is for `./docs/_static/images/` since that separates them from the other `./docs/_static/` files which are also in their own subdirectories.  That leaves room for things like `./docs/_static/images/renesas/` for images that are part of a group.

kisvegabor, 2025-02-20:
It's fine to use the `_static` folder for images. I believe we have the `misc` folder from earlier versinos and engines.

vwheeler63, 2025-02-20:
> It's fine to use the `_static` folder for images. I believe we have the `misc` folder from earlier versinos and engines.

Awesome!  Having done that, and moved the ACTUAL source files to the `./docs/src/` directory, I can confirm that the `./docs/` directory is looking very clean and well-organized now!!  🥳 😃 👍   Now to move the `CODE_OF_CONDUCT.rst` and the translated `README_....rst` files up to the root directory and the clean-up will be complete!

vwheeler63, 2025-03-05:
Resolved by PR #7847