lvgl/lvgl #7805: Translated ./docs/README*.rst Files

vwheeler63, 2025-02-19:
### Introduce the problem

Hi, Gábor!  ( @kisvegabor )

While these translated README files are not inputs for the generated HTML pages at https://docs.lvgl.io/master/ :

- ./docs/README_jp.rst
- ./docs/README_pt_BR.rst
- ./docs/README_zh.rst

I am almost certain I saw them translated somewhere in HTML form, and I currently cannot find them.  (I looked for it at https://lvgl.io/ and https://lvgl.100ask.net/master/index.html .)

Since most `*.rst` files are undergoing a move, I need to find out how these are used and how they are translated to HTML in case my moving them into `./docs/src/` directory will break something that expects them to be directly in the `./docs/` directory.

Looking forward to your reply.

Kind regards,
Vic

### Proposal

Help me know whether it is safe to move these files, or if they should be moved to their own `./docs/` subdirectory since they do not participate in the normal HTML doc-build.

kisvegabor, 2025-02-20:
Hi Vic,

Unused images can be removed without any issues.

CODE_OF_CONDUCT is not really for the docs, but fir developers on GitHub, so we can keep it as Markdown. 

I'm curious of your standpoint, but I think README's for GitHub and shouldn't be part of the docs.

kisvegabor, 2025-02-21:
If we add too many things to the root directory it will be hard to find things there. I don't think that the CODE_OF_CONDUCT is interesting for a wide audience, so it makes sense to not put it to the front. 

I also like the current idea of linking the translated READMEs from the English one. This way they are easy to find but doesn't make the root directory noisy.  

However I do believe the the CODE_OF_CONDUCTS and all RADMEs should be md files.

vwheeler63, 2025-02-21:
> If we add too many things to the root directory it will be hard to find things there. I don't think that the CODE_OF_CONDUCT is interesting for a wide audience, so it makes sense to not put it to the front.

So if I am understanding this right, you want this to remaining in the `./docs/` directory, allowing users to click on it there to read it?

> I also like the current idea of linking the translated READMEs from the English one. This way they are easy to find but doesn't make the root directory noisy.

In case you were not already aware, they already are so linked -- the links appear in the upper-right of the rendered HTML view (from the "Code" tab of the repository).
 
> However I do believe the the CODE_OF_CONDUCTS and all RADMEs should be md files.
>
> What do you think?

Interestingly, I've grown to favor `.rst` files for these reasons:

- they are easier to edit (`.md` files have an issue where to make the text wrap correctly in HTML rendering, the `.md` contents themselves have to NOT be wrapped, which means you have to scroll your editor to the right to read the long lines from your editor.)
- GitHub renders them just as nicely when you click on the file from the "Code" tab of a respository.

But if you feel strongly about it, I'm happy to convert them back to `.md` files.  And just to be clear, from the above discussion it appears you are envisioning all 4 of the above to remain in the `./docs/` directory, like this (i.e. as they are now)?

- ./docs/CODE_OF_CONDUCT.<ext>
- ./docs/README_jp.<ext>
- ./docs/README_pt_BR.<ext>
- ./docs/README_zh.<ext>

Is my understanding correct?

Kind regards,
Vic

kisvegabor, 2025-02-24:
> Is my understanding correct?

Yes, I confirm that. 

I'm still in favor of `md` because it's a more default README format and this way it's easier to keep it in sync with the English README.

closed 2025-03-05 by vwheeler63