lvgl/lvgl #7806: XML Doc-Build Warnings and Errors

vwheeler63, 2025-02-20:
### LVGL version

v9.3

### Platform

Any

### What happened?

Hi, Gábor! ( @kisvegabor )

I see that you did an extensive addition to the documentation under `./docs/details/other-components/xml/`.  I observed when I went to build them and I studied the Sphinx warnings and errors being generated and I noticed there were quite a few coming from these documents.

On any other day I would probably just fix these myself, but I thought this might be an opportunity to "groove you in" on how easy it is to do a local doc-build on your system and then you can see these yourself when you are creating/editing docs in the future.  (I know there are a lot of errors on API docs, and those are still on my radar.)  If you wish to take this on, you can learn some things about `.rst` format and what Docutils (what Sphinx builds on top of) and Sphinx+Breathe like and don't like, and you can do a quick submit and it won't interfere with the re-org I am doing.  

On the other hand, if you're too swamped, I can pick these up as well.

The 2 labeled `CRITICAL` below cause the the EXAMPLES and API sections of `components.rst` get omitted from the HTML output.  All that's wrong is the section headings have underlines that do not correspond to our `./docs/README.md` guidelines.  The error is not with the headings on lines 130 and 135, but all the other ones above line 130.  Once corrected, those warnings will go away.

The C++ parsing errors are covered at the end of the `./docs/README.md`.

Here is the list:

```
C:\TEMP\lvgl_docs\details\other-components\xml\api.rst:61: WARNING: Title underline too short.

``<enumdef>``
************ [docutils]
C:\TEMP\lvgl_docs\details\other-components\xml\api.rst:61: WARNING: Title underline too short.

``<enumdef>``
************ [docutils]
C:\TEMP\lvgl_docs\details\other-components\xml\components.rst:130: CRITICAL: Title level inconsistent:

Example
******* [docutils]
C:\TEMP\lvgl_docs\details\other-components\xml\components.rst:135: CRITICAL: Title level inconsistent:

API
*** [docutils]
C:\TEMP\lvgl_docs\details\other-components\xml\fonts.rst:36: ERROR: Unexpected indentation. [docutils]
C:\TEMP\lvgl_docs\details\other-components\xml\fonts.rst:37: WARNING: Block quote ends without a blank line; unexpected unindent. [docutils]
C:\TEMP\lvgl_docs\details\other-components\xml\fonts.rst:42: ERROR: Unexpected indentation. [docutils]
C:\TEMP\lvgl_docs\details\other-components\xml\fonts.rst:68: WARNING: Unparseable C++ expression: 'lv_style_set_text_font(&style1, <font_name>)'
Error when parsing (type) expression.
If expression:
  Error in postfix expression, expected primary expression or type.
  If primary expression:
    Invalid C++ declaration: Expected identifier in nested name. [error at 32]
      lv_style_set_text_font(&style1, <font_name>)
      --------------------------------^
  If type:
    Invalid C++ declaration: Expected identifier in nested name. [error at 32]
      lv_style_set_text_font(&style1, <font_name>)
      --------------------------------^
If type:
  Error in declarator
  If declarator-id with parameters-and-qualifiers:
    Invalid C++ declaration: Expected identifier in nested name. [error at 23]
      lv_style_set_text_font(&style1, <font_name>)
      -----------------------^
  If parenthesis in noptr-declarator:
    Invalid C++ declaration: Expected ')' in "( ptr-declarator )" [error at 24]
      lv_style_set_text_font(&style1, <font_name>)
      ------------------------^

C:\TEMP\lvgl_docs\details\other-components\xml\images.rst:41: WARNING: Unparseable C++ expression: 'lv_image_set_src(image, <image_name>)'
Error when parsing (type) expression.
If expression:
  Error in postfix expression, expected primary expression or type.
  If primary expression:
    Invalid C++ declaration: Expected identifier in nested name. [error at 24]
      lv_image_set_src(image, <image_name>)
      ------------------------^
  If type:
    Invalid C++ declaration: Expected identifier in nested name. [error at 24]
      lv_image_set_src(image, <image_name>)
      ------------------------^
If type:
  Error in declarator
  If declarator-id with parameters-and-qualifiers:
    Invalid C++ declaration: Expected identifier in nested name. [error at 24]
      lv_image_set_src(image, <image_name>)
      ------------------------^
  If parenthesis in noptr-declarator:
    Invalid C++ declaration: Expected ')' in "( ptr-declarator )" [error at 17]
      lv_image_set_src(image, <image_name>)
      -----------------^

C:\TEMP\lvgl_docs\details\other-components\xml\index.rst:5: WARNING: duplicate label others, other instance in C:\TEMP\lvgl_docs\details\other-components\index.rst
C:\TEMP\lvgl_docs\index.rst:50: WARNING: toctree contains reference to nonexisting document 'API/index' [toc.not_readable]
looking for now-outdated files... none found
pickling environment... done
checking consistency... C:\TEMP\lvgl_docs\details\other-components\xml\translations.rst: WARNING: document isn't included in any toctree [toc.not_included]
```

Kind regards,
Vic

### How to reproduce?

```shell
$ cd ./docs/
$ build.py clean html skip_api
    # Don't pay attention to errors here.
$ touch ./details/other-components/xml/*
    # Tells `build.py` to update and build JUST the XML docs so errors are easier to see.
$ build.py html
    # This will show you JUST the errors on the XML docs.
```

kisvegabor, 2025-02-20:
Thank you, Kevin!

I'll be more strict checking the docs errors next time.

vwheeler63, 2025-02-20:
> Thank you, Kevin!

Are you teasing me?  😆

kisvegabor, 2025-02-21:
Upps, sorry **Vic** 😄 

And thank you for the PR. Yeah, my son is sick, so the family logistics makes effective working a little more difficult. 😅

vwheeler63, 2025-02-21:
> Upps, sorry **Vic** 😄
> 
> And thank you for the PR. Yeah, my son is sick, so the family logistics makes effective working a little more difficult. 😅

No problem at all.  😄

PR #7820 (2025-02-24): docs(xml):  fix Sphinx errors in XML-component docs
Fixes #7806

cc: @kisvegabor 

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.  Done.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.  n/a
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.  n/a
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).  n/a
- R [... cut]
closed 2025-02-24 by kisvegabor