lvgl/lvgl #7821: SVG draw rect off by 1 error

zulusw, 2025-02-21:
### LVGL version

v9.2.0

### Platform

- Linux x86-64 SDL
- Cortex-M33

### What happened?

I have seen a weird issue with SVG rect drawing. It started from this:
`<rect x="192" width="96" height="192" rx="48"/>`
The rectangle above is not drawn correctly (the circles are off by one).
As I tried to play with the dimensions (and looked at the code,) I think the issue resides in `_render_rect` in `lv_svg_render.c`.
AFAIC, the `rc` parameter passed to `lv_vector_path_append` should be initialized like this:
```diff
-lv_area_t rc = {(int32_t)rect->x, (int32_t)rect->y, (int32_t)(rect->x + rect->width), (int32_t)(rect->y + rect->height)};
+lv_area_t rc = {(int32_t)rect->x, (int32_t)rect->y, (int32_t)(rect->x + rect->width - 1), (int32_t)(rect->y + rect->height - 1)};
```

Maybe I'm missing something, but currently, if we define a `1x1` SVG rect we draw 2x2 pixels: is this the expected behavior?

onecoolx, 2025-03-05:
fix this bug at https://github.com/lvgl/lvgl/pull/7888 @zulusw @etag4048

onecoolx, 2025-03-07:
https://github.com/lvgl/lvgl/pull/7902

closed 2025-03-25 by liamHowatt