lvgl/lvgl #7822: Generic MIPI display driver and userdata access

kohlerj, 2025-02-21:
### Introduce the problem

Hi!

First things first, thank you for this amazing library!

**Context**

Currently developing a C++ app to manage filament spool and I thought it would be a great opportunity to start using LVGL. Hardware-wise we are talking about a Pico Pi W and a ST7789 display. 

**The problem**

After giving a look at the codebase I saw that the `user data` field in `lv_display_t` would make a good candidate to store the class instance upon which to call the `send_cmd` and  `send_color` callbacks.

Unfortunately in the case of the ST7789 driver and, by extension, the general MIPI driver, some calls to the callbacks are made before one is given the change to set the `user_data` field.

The only remaining way to define these callbacks is, AFAIK, to use static objects. This is suboptimal in my opinion. 

### Proposal

I was wondering if you guys would be open to splitting up the `create` function into two parts?

- a `create` that would return a `lv_display_t` object 
-  an `init` function that would perform the driver setup with calls the the callbacks

This would allow the user to insert user_data before the callbacks are called. Naming is just a proposition, we can be more inventive and ensure backward compatibility ;)

Side note: How many of us need more than one driver? This would be an argument against performing any change indeed.

liamHowatt, 2025-03-13:
Makes sense, I see the need. Yes, in general the LVGL API never requires the user to use `static` global variables. Could we find a way to do it so that the user can't put it into an invalid state and backwards compatibility is kept? I.e., they forget to call the `init`.

Two ways I can think of:
- The internal `init` happens lazily, at the first flush
- A function `lv_lcd_generic_mipi_create_with_user_data` is added

What do you think?

liamHowatt, 2025-04-15:
Hi, sorry for the delay. In general I think it looks like a good solution. Please open a PR for it. I would recommend making the new function names more self-describing like `..._create_display_no_init` and `..._late_init`.

azais-corentin, 2025-06-17:
Hello, I believe this should be reopened.

I'm having the same issue on the ST7735 driver and it's forcing me to use global, static instances of my classes.

AndreCostaaa, 2025-06-17:
Right. Would you like to pick it up and open a PR for it @azais-corentin ?

azais-corentin, 2025-06-17:
Sure, I'll try coding the proposed changes

axos88, 2026-09-17:
@kohlerj, sorry for lag on getting this fixed. With 9.6 released, and next version being a major release we can now make the proper fix. If the issue is still outstanding on your end, or you have workarounds in place, please take a look if the fixes in #10723 answer your issues

kohlerj, 2026-09-17:
@axos88 Yes! This would definitely fix it.

Thank you for pushing it through!

J

axos88, 2026-10-01:
Hi @kohlerj, the PR is now finished and approved, but it breaks one of our CI processes. We need to give our hardware testing processes some love and do some proper housecleaning to get that green again, which will take some time on our end, but you can take the code as is in the meantime, it will most probably be merged without change in the near future.