lvgl/lvgl #7823: Docs Bug -- Several Bugs Prevent PDF Download Link from Appearing

vwheeler63, 2025-02-22:
### LVGL version

v9.3

### Platform

All

### What happened?

Doc-gen code indicates there should be, in the HTML documentation, a download link for the PDF documentation when it is available.

In ferreting through the code that generates that download link, I found several bugs that was preventing this link from both appearing and working:

- Once PDF was generated, a copy was not placed in the intermediate directory where Sphinx requires it in order to form a download link.
- The programmatically-generated text of the download link was interfering with the top of the top-level `index.rst` file.  What was being done needed a blank line between the link and top text in the `index.rst` file which was a reStructuredText directive, and having no blank line above it was causing the directive to fail.
- The link was being placed on the top-left-hand side of the page, which looked *very out of place* and inappropriate where it was.  It should be on the right side just above the Chinese Translation link.
- The logic that generated the link was in the wrong place in the flow-of-control of the doc-build code in order to work reliably.
- Doc-build flow-of-control was assuming the PDF was directly in the `./docs/` directory *before* the build was started, so if it got generated *during* the doc build, then the PDF would be missing in the intermediate directory where it was needed.

This is what it should look like:

[image]

### How to reproduce?

Run doc-build sequence with PDF present and note the various failures mentioned above.

---

This will be handled in an upcoming PR that refines the documentation structure to be in a more logical sequence and thus makes it easier to use.

lizhaoming634, 2025-02-24:
Have you encountered this issue before?
[image]

kisvegabor, 2025-02-24:
Thank you for investigating it! Does it already on work on your end it yet to be fixed?

vwheeler63, 2025-02-24:
> Thank you for investigating it! Does it already on work on your end it yet to be fixed?

Yes, but there are still a large number of errors generating the `LVGL.pdf` file, mostly Chinese and other characters "...missing from font xxx" that I think should be addressed before releasing the PDF.  So during the doc-gen generation, if the PDF is not present in `./docs/build/pdf/`, then the link will not appear in the HTML output.  What this handles (I have put it in the re-org PR) is that AS SOON AS the PDF-generation errors are addressed, it is just a matter of adding the `latex` argument back into the doc-gen command line alongside the `html` argument and the generated PDF will not only be included in the websites's downloadable files, but the link to it will appear in the HTML output.  Essentially, it eliminates things that were making it NOT WORK, so we're ready for it as soon as the PDF is presentable.

vwheeler63, 2025-03-05:
Fixed by PR #7847

closed 2025-03-05 by vwheeler63