lvgl/lvgl #7833: Keypad/Encoder always requires to be attacted to a group to generate events

recrsn, 2025-02-24:
### LVGL version

v9.2.2

### Platform

CLion on SDL on macOS / esp32s3

### What happened?

Keypad/Encoder always requires to be attached to a group to generate events, this prevents use cases such as custom handling of keyboard events in multiple screens.

This is hardcoded at:

https://github.com/lvgl/lvgl/blob/master/src/indev/lv_indev.c#L770-L771
https://github.com/lvgl/lvgl/blob/master/src/indev/lv_indev.c#L899-L917

There is a workaround available by directly interfacing the hardware, but we like the abstraction provided by lvgl which makes it easier to develop for multiple hardware variants.

### How to reproduce?

```c

int main(void) {
  lv_init();

  // SDL or actual graphics driver on esp32
  lv_display_t *disp = create_display();
  if (disp == nullptr) {
    LV_LOG_ERROR("lv_display_create failed");
    return nullptr;
  }

  static lv_indev_t *indev_encoder = lv_indev_create();
  LV_ASSERT_MALLOC(indev_encoder);
  if (indev_encoder == nullptr) {
    LV_LOG_ERROR("lv_indev_create failed");
    return nullptr;
  }
  lv_indev_set_type(indev_encoder, LV_INDEV_TYPE_ENCODER);

  // omitted for brevity, checks for SDL_GetKeyboardState / actual hardware on esp32
  lv_indev_set_read_cb(indev_encoder, lvgl_read_encoder_cb);
  lv_indev_set_display(indev_encoder, disp);

  lv_indev_add_event_cb(
      indev_encoder,
      [](lv_event_t *event) {
// this does not get called
          const auto manager = static_cast<ScreenManager *>(lv_event_get_user_data(event));
          manager->home();
          lv_event_stop_bubbling(event);
      },
      LV_EVENT_PRESSED, manager);

// rest of hello world code

  while(true)
  {
    ...

    /* Periodically call the lv_timer handler */
    lv_timer_handler();
  }
}
```

kisvegabor, 2025-02-24:
LVGL uses the groups to figure out to which object send the events. Without groups what should be the target widget?

recrsn, 2025-02-26:
> LVGL uses the groups to figure out to which object send the events. Without groups what should be the target widget?

The events generated on input devices do not have a target widget. They simply notify the listeners of lifecycle or an event like button press on the input device.

Like this line here, `hovered` could be `nullptr` too

https://github.com/lvgl/lvgl/blob/master/src/indev/lv_indev.c#L1372

This implementation of `send_event` doesn't really need a target for raising events on input devices

https://github.com/lvgl/lvgl/blob/master/src/indev/lv_indev.c#L1799

First it raises the event on `indev` then dispatches the event to the active widget. For a keypad/encoder without group, it would skip dispatching the event to any widget.

I'll be happy to raise a PR with these changes too.

recrsn, 2025-03-02:
@kisvegabor any thoughts on the above?

recrsn, 2025-03-29:
@lvgl-bot Waiting for feedback from LVGL team

kisvegabor, 2025-04-25:
I'm sorry for the long wait time. 🙁 

I see what you mean now. I agree that even if there is no group we can send still the events via  `lv_indev_send_event`. 

If you are still interested in it a PR would be very welcome! 🙏

recrsn, 2025-04-26:
Hi @kisvegabor

Sure, I'll raise a PR for this

recrsn, 2025-05-23:
I'm working on it in free time, disregard the abandoned status from the bot

kisvegabor, 2025-07-28:
Is it still planned?

closed 2026-02-05 by AndreCostaaa