lvgl/lvgl #7836: Missing commits on/merges to master from changes on release 9.2.x branch

J-u-n-o, 2025-02-24:
### Introduce the problem

Currently missing the bottom border of a label.
I have a clone of lv_port_riverdi_stm32u5 which is using LVGL as a sub-repo.
This is on LVGL commit Commit 62291fa1 'feat(draw): add lv_draw_letter support (#7490)'; somewhere on the master

While investigating I noticed fixes of the 9.2 branch not present on the master.
E.g.  Commit 3a2d1114 `fix(nuttx): fix assert when release LCD draw buffer (#7159)`.

Because this change is about 4 months ago it can be hard to merge to the master.

Suggest to merge fixes (directly/more often) to the master to prevent deviations, and so (possible very hard/difficult) merge conflicts.

kisvegabor, 2025-02-24:
It should be a mistake not a practice. Normally we add fixes to `master` which are merged to the `release` branches if needed. 

@XuNeo could you take a look at this specific commit?

XuNeo, 2025-02-25:
Hi,

The commit `fix(nuttx): fix assert when release LCD draw buffer (#7159)` is intended for release branch only. On master it should be solved in a more proper way. This issue is found during releasing v9.2 when only relevant fixes can be merged.

But it's also true I forgot to open the PR for the more proper fix... 
https://github.com/lvgl/lvgl/pull/7840

J-u-n-o, 2025-02-25:
For me this issue can be closed, but because no `merge lines` from 9.2x to master are present, it triggered me changes/fixes were only present in the 9.2x branch...

XuNeo, 2025-03-12:
Let's close it as completed by. #7840

closed 2025-03-12 by XuNeo