lvgl/lvgl #7837: Color filter cannot be effective on image

yushuailong, 2025-02-24:
### LVGL version

master

### Platform

Eclipse with SDL on Linux

### What happened?

Color filter cannot take effect on image. If `recolor_opa `is not set for `lv_image`, `color_filter `will not take effect.
modify [here](https://github.com/lvgl/lvgl/blob/728546adb7d7bbce71b177132b0c249e4dd47dcc/src/core/lv_obj_draw.c#L218) can solved my problem. But this may not be reasonable

### How to reproduce?

```
 static lv_color_t darken(const lv_color_filter_dsc_t * dsc, lv_color_t color, lv_opa_t opa)
 {
    LV_UNUSED(dsc);
    return lv_color_darken(color, opa);
 }

static void test(void)
{
    static lv_color_filter_dsc_t color_filter;
    lv_color_filter_dsc_init(&color_filter, darken);

    lv_obj_t * avatar = lv_image_create(lv_screen_active());
    LV_IMAGE_DECLARE(img_demo_widgets_avatar);
    lv_image_set_src(avatar, &img_demo_widgets_avatar);
    lv_obj_set_style_color_filter_dsc(avatar, &color_filter, 0);
    lv_obj_set_style_color_filter_opa(avatar, LV_OPA_50, 0);
    lv_obj_center(avatar);
}

![Image](https://github.com/user-attachments/assets/2c16961e-624c-4a41-99ef-0f4c85cafc2e)

```

kisvegabor, 2025-02-24:
It's by design as you need to specify the intensity of recoloring by setting `recolor_opa`. Color filter here adjusts the color to use for recoloring.

yushuailong, 2025-02-25:
> It's by design as you need to specify the intensity of recoloring by setting `recolor_opa`. Color filter here adjusts the color to use for recoloring.

I usually only have the parent widget set `color_filter`, and its child widgets are created externally. In this case, `color_filter` cannot take effect on the image child widgets, and I also need to manually specify the `recolor_opa` of all the image child widgets, which is not a good practice for me.

This example can well illustrate my usage scenario:
``` c
static lv_color_t darken(const lv_color_filter_dsc_t * dsc, lv_color_t color, lv_opa_t opa)
{
    LV_UNUSED(dsc);
    return lv_color_darken(color, opa);
}

static void test(void)
{
    static lv_color_filter_dsc_t color_filter;
    lv_color_filter_dsc_init(&color_filter, darken);
    lv_obj_set_style_color_filter_dsc(lv_screen_active(), &color_filter, 0);
    lv_obj_set_style_color_filter_opa(lv_screen_active(), LV_OPA_80, 0);

    lv_demo_widgets();
}
```
[image]

@kisvegabor  I think `color_filter` should take this scenario into consideration. Do you have any suggestions?

kisvegabor, 2025-02-25:
It's a very creative idea to avoid drawing a screen sized rectangle with half opacity. To be honest I've never though of it. 

I think the key thing here that images doesn't have a color. A button a text or a background have given color which can be modified. However the image is an array without a single color to change. 

Anyway, I see the usefulness of this feature. What we could do is using a special value for `style_image_recolor_opa` (e.g. -1) to indicate that the `recolor_opa` should come from `color_filter_opa`. 

In order to keep the `lv_opa_t` return value for `lv_obj_get_style_image_recolor_opa` I suggest adding `lv_obj_get_style_image_recolor_opa_resolved` which gets the `recolor_opa` first, and if its `-1` gets the `color_filter_opa` instead. 

For the setters we also need ˙lv_style_set_image_recolor_opa_from_color_filter(style, true/false), which internally sets `-1`.

---

The other issue on the image is with the disabled button. To resolved it we need to mix all the color filters of all the parents in `lv_obj_style_apply_color_filter`. For this `color_filter`s shouldn't be inherited but we should get then manually for each parent.

yushuailong, 2025-02-28:
I tried to modify the issue in #7855, there may be some issues, but I verified that it works.

PR #7855 (2025-03-23): feat(refr): add global recolor
Fixes #7837

```c
static void test(void)
{
    lv_obj_set_style_recolor(lv_screen_active(), lv_color_black(), 0);
    lv_obj_set_style_recolor_opa(lv_screen_active(), LV_OPA_80, 0);

    lv_demo_widgets();
}
```
[image]

use layer to cache the global recolor, It can be effective for all widgets.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If y [... cut]
closed 2025-03-04 by yushuailong