lvgl/lvgl #7845: Build error on master: 'lv_draw_unit_t' {aka 'struct _lv_draw_unit_t'} has no member named 'target_layer'

J-u-n-o, 2025-02-25:
### LVGL version

tip of master

### Platform

The commit 84db90f4 'refactor(draw): use task instead of draw_unit as argument to draw functions (#7240)', probably broke the nema build.
The attributes target_layer and clip_area were moved to a different struct.

I am using the lv_port_riverdi_stm32U5 in which recently the stm32 support was added, and was updating the lvgl subrepo to the tip of master...
[lv_port_riverdi_stm32u5](https://github.com/lvgl/lv_port_riverdi_stm32u5)

### What happened?

`
Middlewares/Third_Party/LVGL/lvgl/src/draw/nema_gfx/lv_draw_nema_gfx.c: In function 'nema_gfx_dispatch':
Middlewares/Third_Party/LVGL/lvgl/src/draw/nema_gfx/lv_draw_nema_gfx.c:266:34: error: 'lv_draw_unit_t' {aka 'struct _lv_draw_unit_t'} has no member named 'target_layer'
  266 |     draw_nema_gfx_unit->base_unit.target_layer = layer;
      |                                  ^
Middlewares/Third_Party/LVGL/lvgl/src/draw/nema_gfx/lv_draw_nema_gfx.c:267:34: error: 'lv_draw_unit_t' {aka 'struct _lv_draw_unit_t'} has no member named 'clip_area'
  267 |     draw_nema_gfx_unit->base_unit.clip_area = &t->clip_area;
`

J-u-n-o, 2025-02-25:
In file `lv_draw_nema_gfx.c`
function 
`static void nema_gfx_execute_drawing(lv_draw_nema_gfx_unit_t * u)`
was update,

but next was not:
`static int32_t nema_gfx_dispatch(lv_draw_unit_t * draw_unit, lv_layer_t * layer)`

liamHowatt, 2025-03-12:
Is it fixed after #7759?

J-u-n-o, 2025-04-08:
Using latest updates it is no issue anymore.