lvgl/lvgl #7848: Raw FS Usage

fatihozen3, 2025-02-26:
### Introduce the problem

Hello I use Imxrt1170 mcu and LVGL v9.1.0 .

1. question is , is there any ristriction to usage raw fs ? 
2. question is, may I register 2 file systems into one flash which has different object(different memory base address, different letter etc.)?

I use one rawfs successfully but I can't use 2. one, I don't know why.

liamHowatt, 2025-03-24:
Yes you can use multiple LVGL FS drivers at the same time; however, depending on the driver, there may not be support for more than one instance of the same driver type. Is the "raw fs" driver one you made yourself?

vwheeler63, 2025-03-24:
> I use one rawfs successfully but I can't use 2. one, I don't know why.

Also, in case it is applicable, the use of the "FS identity letter" is not going to seem intuitive for Windows/DOS users.  Have a look at [this documentation](https://docs.lvgl.io/master/details/main-modules/fs.html) if you haven't already for clarification about how it should be used to access more than one file system.

closed 2025-04-23 by lvgl-bot