lvgl/lvgl #7858: #include __LV_TO_STR(LV_CONF_PATH) fails if LV_CONF_PATH contains spaces

XJDHDR, 2025-02-27:
### LVGL version

v9.2.2

### Platform

PlatformIO running through JetBrains CLion on Windows
Custom-made PCB with ESP32-C3 microcontroller

### What happened?

I have a PlatformIO project set up for a project I'm working on. I have my `lv_conf.h` inside the PlatformIO include folder and used this line in the platformio.ini file to tell LVGL where to find it:
`build_flags = -D LV_CONF_PATH="$PROJECT_INCLUDE_DIR/lv_conf.h"`

When I tried to build the project, I encountered this error:
```
In file included from .pio/libdeps/lolin_c3_mini/lvgl/src/lv_init.h:16,
Compiling .pio\build\lolin_c3_mini\src\main.cpp.o
                 from .pio/libdeps/lolin_c3_mini/lvgl/LVGL.h:21,
                 from src/Display/Display.h:8,
                 from src/Display/Display.cpp:5:
.pio/libdeps/lolin_c3_mini/lvgl/src/lv_conf_internal.h:54:42: fatal error: D://College_project/include/lv_conf.h 1/Repoless: No such file or directory
         #include __LV_TO_STR(LV_CONF_PATH)
```

I then tried commenting out the include line mentioned in the error and replaced it with this explicitly defining the file's path:
```
#include "D:/Projects/Repoless Projects/College_project/include/lv_conf.h"
```
This let the project compile successfully.

After undoing that change, I then tried renaming the `Repoless Projects` folder to replace the space with an underscore. This also let my project compile successfully.

XJDHDR, 2025-04-17:
Once again, this problem is still there.

AndreCostaaa, 2025-05-02:
Hello @XJDHDR,
Unfortunately I'm not able to reproduce your error, could I ask you to try this patch ?

```patch
diff --git a/src/lv_conf_internal.h b/src/lv_conf_internal.h
index 0a66d949e..529bad292 100644
--- a/src/lv_conf_internal.h
+++ b/src/lv_conf_internal.h
@@ -49,11 +49,7 @@
 /*If lv_conf.h is not skipped include it*/
 #if !defined(LV_CONF_SKIP) || defined(LV_CONF_PATH)
     #ifdef LV_CONF_PATH                           /*If there is a path defined for lv_conf.h use it*/
-        #define __LV_TO_STR_AUX(x) #x
-        #define __LV_TO_STR(x) __LV_TO_STR_AUX(x)
-        #include __LV_TO_STR(LV_CONF_PATH)
-        #undef __LV_TO_STR_AUX
-        #undef __LV_TO_STR
+        #include LV_CONF_PATH         /* Note: Make sure to define custom CONF_PATH as a string */
     #elif defined(LV_CONF_INCLUDE_SIMPLE)         /*Or simply include lv_conf.h is enabled*/
         #include "lv_conf.h"
     #else
```

XJDHDR, 2025-05-22:
Sorry for taking so long to get back to you. 

I made the change you proposed:
```
    #ifdef LV_CONF_PATH                           /*If there is a path defined for lv_conf.h use it*/
        #include LV_CONF_PATH         /* Note: Make sure to define custom CONF_PATH as a string */
    #elif defined(LV_CONF_INCLUDE_SIMPLE)         /*Or simply include lv_conf.h is enabled*/
        #include "lv_conf.h"
    #else
        #include "../../lv_conf.h"                /*Else assume lv_conf.h is next to the lvgl folder*/
    #endif

```

Sadly, it created a new compile error:
```
In file included from .pio/libdeps/lolin_c3_mini/lvgl/src/lv_init.h:16,
                 from .pio/libdeps/lolin_c3_mini/lvgl/lvgl.h:21,
                 from src/Display/Display.h:14,
                 from src/Display/Display.cpp:16:
.pio/libdeps/lolin_c3_mini/lvgl/src/lv_conf_internal.h:57:12: error: #include expects "FILENAME" or <FILENAME>
   #include LV_CONF_PATH
            ^~~~~~~~~~~~
```

As for you not being able to reproduce it, are you testing with the v9.2.2 release? Maybe you were using the master branch or a different release? I can't think of any other reason you couldn't reproduce it.

> Hello [@XJDHDR](https://github.com/XJDHDR), Unfortunately I'm not able to reproduce your error, could I ask you to try this patch ?
> 
> diff --git a/src/lv_conf_internal.h b/src/lv_conf_internal.h
> index 0a66d949e..529bad292 100644
> --- a/src/lv_conf_internal.h
> +++ b/src/lv_conf_internal.h
> @@ -49,11 +49,7 @@
>  /*If lv_conf.h is not skipped include it*/
>  #if !defined(LV_CONF_SKIP) || defined(LV_CONF_PATH)
>      #ifdef LV_CONF_PATH                           /*If there is a path defined for lv_conf.h use it*/
> -        #define __LV_TO_STR_AUX(x) #x
> -        #define __LV_TO_STR(x) __LV_TO_STR_AUX(x)
> -        #include __LV_TO_STR(LV_CONF_PATH)
> -        #undef __LV_TO_STR_AUX
> -        #undef __LV_TO_STR
> +        #include LV_CONF_PATH         /* Note: Make sure to define custom CONF_PATH as a string */
>      #elif defined(LV_CONF_INCLUDE_SIMPLE)         /*Or simply include lv_conf.h is enabled*/
>          #include "lv_conf.h"
>      #else

AndreCostaaa, 2025-06-10:
Hello everyone. Sorry for taking so much time to get back to this
Could you try this setting the config path with escaped quotes `\"`?

For @XJDHDR :
```
build_flags = -DLV_CONF_PATH="\"$PROJECT_INCLUDE_DIR/lv_conf.h\""
```

For @jgauchia :
```
build_flags = 
  -D LV_CONF_PATH="\"${PROJECT_DIR}/lib/lvgl/lv_conf.h\""
```