lvgl/lvgl #7872: Function doesn't conform to API

kdschlosser, 2025-03-01:
### Introduce the problem

```c
void lv_obj_delete_anim_completed_cb(lv_anim_t * a);
```

### Proposal

That function should be named `lv_anim_delete_completed_cb`. It should also be moved from `src/core/lv_obj_tree.h/c` to `src/misc/lv_anim.h/c`

kdschlosser, 2025-03-01:
@liamHowatt 
@kisvegabor

kisvegabor, 2025-03-04:
Is it because of MicroPython  binding autogenration?

kdschlosser, 2025-03-04:
in order to set the deleted cb you call `void lv_anim_set_deleted_cb(lv_anim_t * a, lv_anim_deleted_cb_t deleted_cb);`

kdschlosser, 2025-03-04:
I think it was overlooked at some point when shuffling around the API.

kisvegabor, 2025-03-25:
Just found this Issue again, sorry. 

> It just doesn't conform to the api. It shouldn't be an lv_obj_ function when the first parameter being passed is lv_anim_t. it should be an lv_anim_ function.

I see, however if it's more like an `anim` function than we need to move it to `lv_anim.c` too, and include all the `lv_obj.h` stuff to delete a widget there. And I'd like to keep `lv_anim` independent of widgets. Basically everything in the `misc` folder are independent of widgets, display, etc.

kdschlosser, 2025-03-26:
I didn't go digging for other anim functions that might not conform to the API. I noticed the one and said.. hey.. why is that like it is... It actually messes up the Micropython binding code generation because the function should become a method of the `lv.anim_t` but it ends up becoming a static method of `lv.obj` which is not exactly where you would think it should be....

kdschlosser, 2025-03-26:
There are other things that are not really where they should be ...

These should have the prefix `lv_line_` instead of `lv_obj_`
```
lv_obj_get_style_line_width
lv_obj_get_style_line_dash_width
lv_obj_get_style_line_dash_gap
lv_obj_get_style_line_rounded
lv_obj_get_style_line_color
lv_obj_get_style_line_color_filtered
lv_obj_get_style_line_opa

lv_obj_set_style_line_width
lv_obj_set_style_line_dash_width
lv_obj_set_style_line_dash_gap
lv_obj_set_style_line_rounded
lv_obj_set_style_line_color
lv_obj_set_style_line_color_filtered
lv_obj_set_style_line_opa
```

These should be `lv_image_`
```
lv_obj_get_style_image_opa
lv_obj_get_style_image_recolor
lv_obj_get_style_image_recolor_opa
lv_obj_set_style_image_opa
lv_obj_set_style_image_recolor
lv_obj_set_style_image_recolor_opa

```

and these should be `lv_arc_`
```
lv_obj_get_style_arc_width
lv_obj_get_style_arc_rounded
lv_obj_get_style_arc_color
lv_obj_get_style_arc_opa
lv_obj_get_style_arc_image_src
lv_obj_set_style_arc_width
lv_obj_set_style_arc_rounded
lv_obj_set_style_arc_color
lv_obj_set_style_arc_opa
lv_obj_set_style_arc_image_src
```

these should be `lv_label_`
```
lv_obj_get_style_text_color
lv_obj_get_style_text_opa
lv_obj_get_style_text_font
lv_obj_get_style_text_letter_space
lv_obj_get_style_text_line_space
lv_obj_get_style_text_decor
lv_obj_get_style_text_align

lv_obj_set_style_text_color
lv_obj_set_style_text_opa
lv_obj_set_style_text_font
lv_obj_set_style_text_letter_space
lv_obj_set_style_text_line_space
lv_obj_set_style_text_decor
lv_obj_set_style_text_align
```

closed 2025-04-23 by lvgl-bot