lvgl/lvgl #7884: Where to go for help when LVGL fails?

klaatu2a, 2025-03-04:
### Introduce the problem

When there is a problem, where do you go for LVGL help?   
It would seem that the LVGL forum https://forum.lvgl.io/ would be the logical goto, but there is no real trouble shooting category.
Posts on Squareline Studio's forum go unanswered.
Is this Github discussion the place to go?  Is it actively supported by LVGL staff?

vwheeler63, 2025-03-05:
> Is it actively supported by LVGL staff?

It is indeed.  What are you running into?

klaatu2a, 2025-03-05:
Well, I'll just post it and maybe it means something to you.  

This is LVGL 9.2.2
ESP-IDF forced to LVGL 9.2.2
ESP32-S3-LCD-EV-BOARD v1.1

Guru Meditation Error: Core 0 panic’ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC : 0x4201b50f PS : 0x00060830 A0 : 0x8201c42d A1 : 0x3fcb8570
— 0x4201b50f: lv_draw_add_task at E:/path/managed_components/lvgl__lvgl/src/draw/lv_draw.c:95
…
— Error: ClearCommError failed (PermissionError(13, ‘Access is denied.’, None, 5))

The offending line 95 is below:

90l :v_draw_task_t * lv_draw_add_task(lv_layer_t * layer, const lv_area_t * coords)
91:{
92:    LV_PROFILER_BEGIN;
93:    lv_draw_task_t * new_task = lv_malloc_zeroed(sizeof(lv_draw_task_t));
94:
95:    new_task->area = *coords;
96:    new_task->_real_area = *coords;
    new_task->clip_area = layer->_clip_area;

klaatu2a, 2025-03-05:
Why not use the LVGL forum rather than this message board?   The LVGL forum is more feature reach.  For example, I can post an image inline.

vwheeler63, 2025-03-06:
> For example, I can post an image inline.

You can post images here as well.  One of the easiest ways is to simply paste it from your clipboard while keyboard focus is in the editing area.

About the above -- I am thinking we have some multi-core ESP experts available to consult with.  @kisvegabor Gábor, do you know who would be best to interpret the above?

klaatu2a, 2025-03-06:
I don't believe the 2nd core is being used.

vwheeler63, 2025-03-06:
> I don't believe the 2nd core is being used, but maybe so

Okay.  Just to be clear, is this message being issued from within an IDE?

klaatu2a, 2025-03-07:
This is a run time error as output by the code

[image]

klaatu2a, 2025-03-07:
This seems somewhat insurmountable.   Debugging code that the was machine generated and code which I have no idea what its trying to accomplish, nor am I familiar with the function or elements thereof.
I added a few print statements just to get a sense for what is happening.  Its not a hard fault, but rather it repeatedly calls this function before something happens which causes the permissions error.  It does fail at the same iteration each time

CODE

[image]

Run Time output:

[image]

vwheeler63, 2025-03-08:
Hi, @klaatu2a !  This is excellent info.  @kisvegabor has the best knowledge of the drawing pipeline, and he may be delayed since it is on a business trip, but should be back in a matter of a few days.

AndreCostaaa, 2025-03-09:
Seems like a classic out of memory error 

Could you check for the value of `new_task` after the call to `lv_malloc_zeroed` in line 93?

Something like

```c
lv_draw_task_t * new_task = lv_malloc_zeroed(sizeof(lv_draw_task_t));
if(!new_task) {
    printf("New Task is Null\n"); 
    while (1) ;
}
```

klaatu2a, 2025-03-09:
[image]

klaatu2a, 2025-03-09:
There is no lv_conf.h in the project folders.  Instead I find an entry in the SDKconfig, it is set to 64K.  Which is much greater than the 2KB minimum.  I changed it to 128K,
No change, still failed

[image]

klaatu2a, 2025-03-09:
Wait, I did not save the change to 128K.   With this change and removing all the print statement, that issue is gone...  Now its failing in a new way.   Trouble shooting this now

klaatu2a, 2025-03-09:
Watchdog!

[image]

Since this is a conversion from LVGL 8.4 to LVGL 9.2.2, I wonder if there is something vestigial in the SDKconfig that is impairing LVGL 9.2.2

AndreCostaaa, 2025-03-09:
Could you show me the code calling `lv_timer_handler` please?

I'm wondering if you're calling that function in a infinite loop without adding a small delay between each call

klaatu2a, 2025-03-09:
I disabled the watchdog, it is spinning somewhere forever...blank screen

I do not have an file with the name lv_timer_handler

AndreCostaaa, 2025-03-12:
Could you try increasing `LV_LOG_LEVEL`. In the same place you modified `LV_MEM_SIZE`

Try modifying it to `LV_LOG_LEVEL_TRACE` 

Currently having the same problem as you in a completely different platform so maybe my problem will be the same as yours

klaatu2a, 2025-03-13:
The code was generated with Squareline Studio.  The only response I got from them was that they are not compatible with ESP-IDE v5.4.  No details at all.

klaatu2a, 2025-03-13:
If I set it to the most verbose, it hits a stack overflow.
Tuning it down to the next level:
it prints forever indicating no memory.  I increased the "Size of the memory used by `lv_malloc()` in kilobytes" from 96K to 128K, same issue.  

[Info]  (43.510, +5)     lv_malloc: used: 82808 ( 92 %), frag:   1 %, biggest free: 8064 lv_mem.c:77
[Warn]  (43.510, +0)     lv_draw_buf_create_ex: No memory: 157x24, cf: 16, stride: 628, 15072Byte,  lv_draw_buf.c:296
[Warn]  (43.510, +0)     lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:442
[Info]  (43.510, +0)     lv_malloc: couldn't allocate memory (283911 bytes) lv_mem.c:73
[Info]  (43.510, +0)     lv_malloc: used: 82808 ( 92 %), frag:   1 %, biggest free: 8064 lv_mem.c:77
[Warn]  (43.510, +0)     lv_draw_buf_create_ex: No memory: 177x401, cf: 16, stride: 708, 283908Byte,  lv_draw_buf.c:296
[Warn]  (43.510, +0)     lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:442
[Info]  (43.510, +0)     lv_malloc: couldn't allocate memory (333743 bytes) lv_mem.c:73
[Info]  (43.510, +0)     lv_malloc: used: 82808 ( 92 %), frag:   1 %, biggest free: 8064 lv_mem.c:77
[Warn]  (43.510, +0)     lv_draw_buf_create_ex: No memory: 205x407, cf: 16, stride: 820, 333740Byte,  lv_draw_buf.c:296
[Warn]  (43.515, +5)     lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:442
[Info]  (43.515, +0)     lv_malloc: couldn't allocate memory (15075 bytes) lv_mem.c:73
[Info]  (43.515, +0)     lv_malloc: used: 82808 ( 92 %), frag:   1 %, biggest free: 8064 lv_mem.c:77
[Warn]  (43.515, +0)     lv_draw_buf_create_ex: No memory: 157x24, cf: 16, stride: 628, 15072Byte,  lv_draw_buf.c:296
[Warn]  (43.515, +0)     lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:442
[Info]  (43.515, +0)     lv_malloc: couldn't allocate memory (283911 bytes) lv_mem.c:73
[Info]  (43.515, +0)     lv_malloc: used: 82808 ( 92 %), frag:   1 %, biggest free: 8064 lv_mem.c:77
[Warn]  (43.515, +0)     lv_draw_buf_create_ex: No memory: 177x401, cf: 16, stride: 708, 283908Byte,  lv_draw_buf.c:296
[Warn]  (43.515, +0)     lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:442
[Info]  (43.515, +0)     lv_malloc: couldn't allocate memory (333743 bytes) lv_mem.c:73
[Info]  (43.520, +5)     lv_malloc: used: 82808 ( 92 %), frag:   1 %, biggest free: 8064 lv_mem.c:77
[Warn]  (43.520, +0)     lv_draw_buf_create_ex: No memory: 205x407, cf: 16, stride: 820, 333740Byte,  lv_draw_buf.c:296
[Warn]  (43.520, +0)     lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:442
[Info]  (43.520, +0)     lv_malloc: couldn't allocate memory (15075 bytes) lv_mem.c:73
[Info]  (43.520, +0)     lv_malloc: used: 82808 ( 92 %), frag:   1 %, biggest free: 8064 lv_mem.c:77
[Warn]  (43.520, +0)     lv_draw_buf_create_ex: No memory: 157x24, cf: 16, stride: 628, 15072Byte,  lv_draw_buf.c:296
[Warn]  (43.520, +0)     lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:442
[Info]  (43.520, +0)     lv_malloc: couldn't allocate memory (283911 bytes) lv_mem.c:73
[Info]  (43.525, +5)     lv_malloc: used: 82808 ( 92 %), frag:   1 %, biggest free: 8064 lv_mem.c:77
[Warn]  (43.525, +0)     lv_draw_buf_create_ex: No memory: 177x401, cf: 16, stride: 708, 283908Byte,  lv_draw_buf.c:296

klaatu2a, 2025-03-13:
I put a breakpoint at that "lv_malloc: couldn't allocate memory:" print statement.  
It appears to fail right from the start:

[image]

klaatu2a, 2025-03-13:
Note the variables of draw_buf_malloc
The color format should be RGB565, not RGB888

[image]

liamHowatt, 2025-03-14:
> Note that it is configured to "No OS", so LVGL handles memory

It is actually `LV_USE_STDLIB_MALLOC` that decides who handles memory. It will be handled by LVGL if it is set to `LV_STDLIB_BUILTIN`.

klaatu2a, 2025-03-14:
Can't find any setting "LV_USE_STDLIB_MALLOC"
But this is in the LVGL project settings:
[image]

klaatu2a, 2025-03-14:
Changed the Malloc from LVGL built-in, to that shown below
And its now working.  Its not rendering all the elements correctly, but it 95% there

[image]

closed 2025-03-31 by klaatu2a